sandeepmistry/arduino-nRF5 #183: Add Black Magic Probe (BMP) upload method

dmikhalsky, 2017-08-10:
The method is already implemented in Roger Clark's fork of RBL rf51822 cores so it's just a matter of merging two boards.txt files.

sandeepmistry, 2017-08-11:
@dmikhalsky I don't have a Black Magic Probe to test with, please open a pull request with the minimal changes required add support to continue the discussion.

ntavish, 2017-08-11:
Hi @dmikhalsky , I have added basic support for upload using BMP here https://github.com/electronut/arduino-bluey/commit/1fb4e5a14a744a3bda581f8888f607e7347969ef

But I was not able to figure out a way to make it work in a way that would be compatible with how it done done with openocd in this repo. One issue is flashing the softdevice, which uses the `bootloader.tool` setting, which I had to set to gdb (instead of openocd over here), so it disregards actual programmer and always uses gdb and the serial port selected (as the BMP serial port).

micooke, 2017-08-20:
I've added bmp locally by modifying platform and programmers.

@ntavish - bmp uses gdb, not openocd. Luckily this is included in sandeeps repo!

I need to tidy up the gdb entry point, but its worked great for me. One thing ive implemented is that if the soft device has been downloaded, 'burn bootloader' will upload the soft device. I can't remember if this is already implemented for the other programming devices?

Ill try and put on a pull request in the next day or so.

sandeepmistry, 2017-11-21:
Closed via #196. @micooke thanks!

closed 2017-11-21 by sandeepmistry