sandeepmistry/arduino-nRF5 #187: "java.lang.NullPointerException " while trying to upload the code for BLE 400

anilkunchalaece, 2017-08-21:
Hi,

I am try to upload the code to the[ BLE 400](http://www.waveshare.com/wiki/BLE400) from waveshare with library

I installed the cp210x drivers (windows 10 64 bit) and it is showing com port also

It is uploading successfully but when i try to upload the code to the board it is showing following error

```
java.lang.NullPointerException
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingProgrammer(SerialUploader.java:314)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:89)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.SketchController.upload(SketchController.java:713)
	at processing.app.SketchController.exportApplet(SketchController.java:686)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2149)
	at java.lang.Thread.run(Thread.java:745)
```
How to avoid this error ?

dlabun, 2017-08-21:
When you go to upload the sketch do you have the COM port selected? Also, please let us know what version of the IDE you are using.

anilkunchalaece, 2017-08-21:
Hi, Thanks for Quick Reply.

I selected the COM5, it is showing in the Device Manager also.
 arduino version is 1.8.3

dlabun, 2017-08-21:
It doesn't look like that board includes a programmer on it, that UART to USB adapter is only for looking at the serial port on the nRF chip. Do you have an external programmer you can use?

anilkunchalaece, 2017-08-21:
I have nrf51-dk

micooke, 2017-08-21:
@dlabun is correct - i have this board. Its attached to the serial pins, it is not a swd programmer.

You need to buy an stlink or equivalent to program the nrf5 series, and attach it to the programming pins swdio swdclk

dlabun, 2017-08-28:
Closing, no activity

closed 2017-08-28 by dlabun