sandeepmistry/arduino-nRF5 #189: Support for both Hardware I2C (i.e. Wire1)

micooke, 2017-08-24:
Im wondering if there is a reason that only one of the two hardware I2C devices are currently exposed?
Are there conflicts that i am unaware of? It looks like Wire1 was exposed at some point

Ive enabled them both locally, but i havent tested both at the same time.

The change requires modification of Wire.h, Wire_nRF51.h, Wire_nRF52.h and any board variant.h that requires it.

micooke, 2017-08-24:
Okay i think I'm going to answer my own question - it looks like SPI and I2C share the same interrupt? This would make sense why Wire uses the interrupt for Wire1.

I had an example where I enabled both SPI, then tried to use an SPI oled which caused the SPI to hang (it dropped into an infinite loop)

closed 2017-08-24 by micooke

sandeepmistry, 2017-09-05:
@micooke if there's demand for 2 Wire instances we could add it, https://github.com/sandeepmistry/arduino-nRF5/pull/177/files added support for 2 SPI devices.

We could make the ISR handler weak.

micooke, 2017-09-05:
That would be great! Sorry, i missed that issue/PR.
The smart watches have two I2C devices on separate pins, at the moment im using [Arduino_Software_I2C](https://github.com/Seeed-Studio/Arduino_Software_I2C) for both as i have an undefined issue with the hardware Wire library that isnt there for the software library