sandeepmistry/arduino-nRF5 #190: Low Power NUS UART bridge

kriswiner, 2017-08-25:
I have been struggling with lowering the average power of nRF52 applications within the confines of this Arduino core and BLEPeripheral. I tried ArduinoLowPower but had mixed results, partly because of the interplay between the Arduino core, BLEPeripheral and the LowPower API. There really needs to be an integrated approach, something like Adafruit has [done](https://github.com/adafruit/Adafruit_nRF52_Arduino), perhaps, but that core really has no low power options either.

I decided to just start simply, and I have managed to use the Nordic UART service example from BLESerial.h to test what kinds of power savings are possible. This is the sketch I came up with, loaded onto an nRF52 add-on board acting as a BLE UART bridge from an STM32L4 development board to a smartphone console:

     // Import libraries (BLEPeripheral depends on SPI)
     #include <SPI.h>
     #include <BLEPeripheral.h>
     #include "BLESerial.h"

     // define pins (varies per shield/board)
     #define BLE_REQ   10
     #define BLE_RDY   2
     #define BLE_RST   9

     // create ble serial instance, see pinouts above
     BLESerial BLESerial(BLE_REQ, BLE_RDY, BLE_RST);

     void setup() {
        // custom services and characteristics can be added as well
       BLESerial.setLocalName("UART");

       Serial.begin(115200);
       BLESerial.begin();

       // acceptable values are: -40, -30, -20, -16, -12, -8, -4, 0, 4
       int power = -4;
       sd_ble_gap_tx_power_set(power);

       }

     void loop() {
       BLESerial.poll();

        __delay(1000);

       send();

       sd_power_mode_set(NRF_POWER_MODE_LOWPWR);
      sd_app_evt_wait();
      }

     void __delay(uint32_t timeout)
     {
     uint32_t start;
   start = millis();

   do
   {
       __WFE();
   }
   while ((millis() - start) >= timeout);
  }

   void send() {
  if (Serial) {
    int byte;
    while ((byte = Serial.read()) > 0) BLESerial.write((char)byte);
  }
  }

Without the sd-calls for low power and the special delay function, the serial bridge takes ~6.0 mA. With the low power calls and the special delay, the bridge requires ~4.0 mA. I believe the Serial peripheral takes ~2 mA all by itself, but this is unavoidable ina serial bridge application. The TX Power functionality works, but I notice hardly any difference in the average power whether the TX Power is at -40 or -4 (default) dB. But the RSSI sinks like a stop at lower TX power levels, so my advice is to keep it at -4 dB and not try to use this as a power savings method.

The last item that will matter is the advertising interval. I notice in another application using these methods that sensor data is updated many times a second, which is not really necessary. I could widen the reporting threshold for changes, but I would like to find a method to manage the TX interval directly.

There is likely some soft device API sd_... that does this and that we can use for low power appkications.

dlabun, 2017-08-28:
Seems like a duplicate of #168 , Can you move this information over to that post?

kriswiner, 2017-08-28:
Hardly, I am explaining how one can save 4 mA out of 6 mA without using the
ArduinoLowPower API. Unfortunately there is no forum for nRF52 Arduino
users, so this is FYI not an issue per se. You may move it or delete it as
you like.

dlabun, 2017-08-28:
That's totally fine, it just wasn't clear to me if it was the same thing. You may want to clarify in the title that ArduinoLowPower API is not involved.

curtpw, 2017-08-29:
I support any and all discussions of power management for this project. I work with wearable devices at a mental health research institute in NYC. Using Arduino code is great for open science because it makes it easier to share code with other researchers - which makes it easier to reproduce and verify research. It also makes it easier for someone like me, with a background in ChemE and no background in EE, to build devices quickly and cheaply. **Power management is the most significant issue holding back the nRF5X Arduino Core from being a tool for rapid prototyping of devices in a research setting.** In my experience this project is superior to mbed, primo or any other non-Nordic SDK approach to the nRF52832, but until power management is included it will be difficult to use outside of hobby/tinkering applications.

kriswiner, 2017-08-29:
Well, with the above tricks we are at 1.25 mA average power usage, which is
a nice start. TX power and advertising interval don't matter much in terms
of power usage. Having serial and I2C peripherals enabled does, so we am
working on allowing these to be disable and let the nRF52 enter sleep mode
properly, where it will consume ~20 uA instead of the 1.2 mA it does now
when almost everything prevents the soft device from taking advantage of
the lowest power modes. But the I2C API needs to be changed to accomodate
this, etc. The ArduinoLowPower API won't help here. But a few simple
changes to the Arduino core might...

curtpw, 2017-08-29:
Awesome, thanks for working on this. 

What do you think of enabling the DC/DC converter on the nRF52 ?
Sort of like this https://www.thisisant.com/developer/resources/tech-bulletin/this-one-line-of-code-could-halve-your-power-consumption-on-d52-and-n5-modu/

kriswiner, 2017-08-29:
> bulletin/this-one-line-of-code-could-halve-your-power-
> consumption-on-d52-and-n5-modu/

The DC/DC converter has to be included in the circuit. For a lot of devices
it isn't. Just like the LSE can save power too but a lot of devices don't
bother adding a 32kHz crystal.

I don't think these are the big players anyway. Documentation shows that if
you include a DC/DC inductor and extra cap you can save 25% on peak
current. But UART takes 1.2 - 1.3 mA even when idle.

Proper peripheral management (not adequately addressed in the Arduino core)
is the low hanging fruit at the moment.

sandeepmistry, 2017-09-05:
> You may move it or delete it as you like.

Let's close this for now, but feel free to continue the discussion. I'm surprised the Serial is the root cause, calling Serial.end() or not calling Serial.begin(...) will disable it.

closed 2017-09-05 by sandeepmistry

curtpw, 2017-10-03:
Has there been progress on managing I2C and SPI peripherals to decrease power consumption? Would be quite useful....

kriswiner, 2017-10-03:
I managed to get the power usage down to ~1 mA using low power mode and
__WFE.