sandeepmistry/arduino-nRF5 #198: NRF52 PDK (Nrf52840 board support)

rsramki, 2017-10-03:
I am new to NRF52 boards. Unable to get this working with NRF52 PDK board (PA100056)   run and program my new PDK board after downloading Arudino Latest IDE and add this json in board manager.

https://www.nordicsemi.com/eng/Products/nRF52840-Preview-DK

dlabun, 2017-10-03:
Do you have the nRF52 DK or nRF52840 Preview DK? They're two different boards with different chips. Only the nRF52 DK has been added so far.

rsramki, 2017-10-03:
nRF52840 Preview DK, how can we get this enabled. willing to help

d00616, 2017-10-06:
@rsramki There is another issue with this topic: https://github.com/sandeepmistry/arduino-nRF5/issues/70

The nRF52840 introduces multiple GPIO ports. In my opinion the pin configuration mechanism has to be changed. For portability the Arduino variant of board description is a good variant to start with: https://github.com/arduino-org/arduino-core-nrf52/blob/master/variants/arduino_primo/variant.cpp

dlabun, 2017-10-06:
@d00616 hit it right on the head, we already have an open issue for adding that board. So far nobody has done a PR for that board, that would be next step to get it support.

rabbithat, 2017-10-14:
Looks like there is an nRF52840 "variant": https://github.com/lpercifield/arduino-nRF5/commit/d55d54a1bdc479acc259e131f0f445c5da8e02b3
Would that work? If so, how exactly should it be added so that it will appear in the Arduino IDE board manager list of boards?

rabbithat, 2017-10-16:
Is there any ETA on when we are likely to see Nrf52840 board support?  Are we talking 2Q2018 or later?

dlabun, 2017-10-16:
There's currently no ETA, somebody from the community would need to submit a PR with the appropriate changes.

dlabun, 2017-10-16:
Closing as a duplicate of #70

closed 2017-10-16 by dlabun