sandeepmistry/arduino-nRF5 #199: I2C freeze. Custom NRF52

snowneki4, 2017-10-03:
I have a custom nrf52 board and i need to get MPU6050 running. I have connected it using following pins:
#define PIN_WIRE_SDA         (3u)
#define PIN_WIRE_SCL         (21u)
Both pins have 4.7k resistors connected.
I have not yet implemented MPU6050 library, though i am trying to recognize the device using I2C scanner described here: [https://playground.arduino.cc/Main/I2cScanner](url) Code i am using is there.
Whenever sensor is disconnected terminal continues to show No I2C devices found. Which is right. But whenever i connect sensor to the pins it freezes after the scanning stage. 
So, questions. Originally pins for SDA/SCL were 20/21 in variants.h. Can i actually change those? 
Can you point me what am i doing wrong? 

My skills arent very high, so could be i am doing something completely stupid.

micooke, 2017-10-05:
See #176

lichtheini, 2017-10-05:
Maybe I can help:  I have an Adafruit Feather nRF52 and a MPU9250, so much the same condition. There are many points were your code or wiring could be wrong. I just show the one, that took me a while to figure out: a hanging I2C-bus with SDA low. 

My solution, well, my workaround is described  here: [https://github.com/esp8266/Arduino/issues/1025#issuecomment-319199959](https://github.com/esp8266/Arduino/issues/1025#issuecomment-319199959) and [here](http://www.forward.com.au/pfod/ArduinoProgramming/I2C_ClearBus/index.html).

```
if(digitalRead(PIN_WIRE_SCL) == HIGH && digitalRead(PIN_WIRE_SDA) == LOW) {
      //Serial.println("reset i2c bus");
      pinMode(28, OUTPUT);  // connect parallel to SCL
      digitalWrite(28, LOW);
      delay(200); 
      pinMode(28, INPUT);
      delay(50);
      NVIC_SystemReset();
}
```

And yes, almost any pins are usable for I2C communication. 

> The GPIOs used for each two-wire interface line can be chosen from any GPIO on the device and are
independently configurable.  (Product Spec  rev 1.3, p. 305)

Hope it helps!

snowneki4, 2017-10-08:
Problem was somehow solved my selecting Generic NRF51 option from the board selection menu, even
though i am using nrf52 board.
It is described in the post micooke has mentioned.
thanks everyone

closed 2017-10-08 by snowneki4

micooke, 2017-10-08:
Glad it worked for you!
I'm sure this will be resolved in the near future for the nrf52. Im not sure if it helps, but you could also try this : https://github.com/micooke/SoftwareI2C

snowneki4, 2017-10-08:
Will give it a try as well.

micooke, 2017-10-08:
No problem. I just realised its a little outdated (but should still work), ill try and push the latest revision tomorrow. 
Only API changes i made are that sda, scl were removed from begin() and put into the constructor.