sandeepmistry/arduino-nRF5 #200: Taida_Century_nRF52_minidev - pin mapping

micooke, 2017-10-05:
First up - the README reference no longer has a valid link. Im planning on hosting the schematics i was provided, so i can provide somewhere to point to if needed. For reference, it seems to also be called 'Gold Core' at this point in time on aliexpress.

Regarding the variant.h pin mapping by @carlosperate and @d00616, was the pin mapping for tx, rx, i2c and spi arbitrarily chosen?

I ask because the 12 pin header exposes tx and rx, which are connected to either;
* GPIO3(Tx),  GPIO2(Rx) if J11, J12 in RGZ are soldered
* GPIO2(Tx),  GPIO1(Rx) if J15, J16 in RSM are soldered

I currently have RGZ soldered, but either option creates a conflict with the SDA, SCL pin mapping.

Is there a relationship between the pin allocation of TDI/TDO and MOSI/MISO on the nRF52? If so then one of RGZ or RSM may be preferable as the remaining two jumpers on the reverse side map TDI/TDO.

dlabun, 2017-10-05:
We probably shouldn't have been linking to AliExpress in the first place, but then again I can't find a manufacturer page for these modules.

carlosperate, 2017-10-05:
Yeah, I don't see a better alternative either than the online shop product pages. It'd be great if somebody could archive the data somewhere a bit more dependable, so that would be quite helpful.

About the pin out, not sure where it originated from, @d00616 would be better position to know that answer. It's possible that similar modules might have slightly different pin outs, @micooke are you certain your module was the same @micooke had when https://github.com/sandeepmistry/arduino-nRF5/pull/47 was introduced?

micooke, 2017-10-05:
I agree, and no i haven't found a manufacturer either. This is why i might just put up some info and schematic in my github.io repo for archival purposes.

The schematic i have claims its a 'CC2640RSM/RGZ Module Test Board V1.0' from 'rf-smart.taobao.com', which looks like another store.

micooke, 2017-10-05:
Hi @carlosperate - im not 100% sure, but if i google 'taida century nrf52832' the first result is an aliexpress link (that is no longer available) which looks identical to the board i have, down to the silk screen. 

If you search aliexpress for 'nrf52832 gold core' (2 results only) that's what i have. It is a nrf52832 module soldered to a 'CC2640 test module' (aliexpress - 1 result), which matches a schematic i have (and will host and post tomorrow). I have seen modules for the CC2640, nrf52832 and nrf51822 that match this test module/breakout board.

The 2xbutton and 2xled bindings are identical to the Taida pin mapping.

micooke, 2017-10-06:
See https://micooke.github.io/nRF52832_TaidaCentury_GoldCore
Files are located here : https://github.com/micooke/micooke.github.io/tree/master/nRF52832_TaidaCentury_GoldCore

d00616, 2017-10-06:
The reason for the layout is, that if haven't realized the existence of the jumpers. My documentation is at https://github.com/RIOT-OS/RIOT/wiki/Board%3A-nRF52-minidev

micooke, 2017-10-09:
@dlabun - I was thinking that i can accomodate for the original pin mapping, plus the pin mapping enabled by soldering the RGZ or RSM configuration by using board_variant.

The changes i have made locally are a bit of a hack. They set compile-time defines using ```{build.lfclk_flags}``` (as there is only one static option for this), based off the ```board_variant```, which are used in variant.h to change the pin mapping.

I couldn't use ```extra_flags``` as this is set by the softdevice chosen, and there are two options.