sandeepmistry/arduino-nRF5 #201: [Question] Using arduino primo examples with nRF52 DK board

mitko9000, 2017-10-05:
Hello guys i have two nRF52 DK laying around and I was wondering can I upload an Arduino Primo code to them ?

I can successfully use nRF52 DK with bluetooth peripheral with S132 soft device testing the bluetooth and the blink program

dlabun, 2017-10-05:
Maybe... The Primo has a service processor on it so as long as you aren't using any of those functions you should be able to use your code on both boards.

mitko9000, 2017-10-06:
@dlabun Any pointers how to do this easily because I have to somehow transfer primo libraries code to the nrf52 core here ?

Or this is not done yet

dlabun, 2017-10-06:
It all depends on your code so it's hard for me to say... Right off the bat, anything like the WiFi or IrRD features or the ON, BLE or User2 LEDs can't be used with the nRF52 DK.

dmikhalsky, 2017-10-12:
I did it. Successufully. 
You have to install Primo board in Arduino IDE and compile for it. Then, it generates a combined "SD + bootloader + code" hex file. You simply drag the file to Segger drive and voila, it works.

mitko9000, 2017-10-12:
@dmikhalsky How to compile without having the board ?

dmikhalsky, 2017-10-12:
??? As I said, just install it via the Board manager, choose "Arduino Primo" as your board, load any example and press "Upload"
Of course, you'll get an error, but there will be a generated hex file.

mitko9000, 2017-10-12:
@dmikhalsky Ok I got that there was a option in the preferences for arduino to show me the compilation path dont have a board right now to try but I will try it ASAP.

Thanks 👍

dmikhalsky, 2017-10-12:
Cool. If you need Serial (of course, you need :-), you'll also have to change pin definitions in variants.c, but that's another story.

mitko9000, 2017-10-12:
@dmikhalsky As far as I checked it should be trivial to replace the variants.c with the nrf52 dk kit from here its just simple pinout :)

dmikhalsky, 2017-10-12:
Didn't work for me, but might work for you

closed 2017-11-21 by dlabun