sandeepmistry/arduino-nRF5 #204: Calliope mini support?

joernalraun, 2017-10-16:
Hi, it would be cool to have additional support for the Calliope mini (a board based on the microbit with a different shape, more internal sensors, Grove connectors, RGB LED and a motordriver).
There is a different motion sensor (BMX055) included and a slightly different pin layout:
https://calliope-mini.github.io/v10/

sandeepmistry, 2017-11-13:
Hi @joernalraun,

I don't have access to this board, please provide a pull request if you would like support. You can use https://github.com/sandeepmistry/arduino-nRF5/commit/7bbfca6bad0ec88abbffcd109ad7e68b444ccf25#diff-a2d72f9d913ff95fe85624bf8f89865a as a reference.

sandeepmistry, 2018-03-18:
Closed via #259 by @jhmaloney

closed 2018-03-18 by sandeepmistry