sandeepmistry/arduino-nRF5 #206: nRF52 - use NFC pins as GPIO

micooke, 2017-10-17:
I know this adds yet another compile and variant option, but for nRF52 boards PO.09 and PO.10 are by default reserved for a NFC antenna. Are we able to support an option under variants to enable/disable this option and hence free up two additional pins if you aren't using NFC?

To use these pins as GPIO you need to #define CONFIG_NFCT_PINS_AS_GPIOS

micooke, 2017-10-17:
To add more to this, as mentioned in a previous issue (that i cant find) it would be nice to add an option to enable/disable PO.29 as a reset pin via #define CONFIG_GPIO_AS_PINRESET

dlabun, 2017-10-17:
For the reset, check PR #91 as there's already some discussion there.

micooke, 2017-10-17:
Maybe we can bundle them into one variant selection field with four options? They are both GPIO configuration options.

dlabun, 2017-10-20:
Not a bad idea at all

micooke, 2017-10-20:
I have this working locally, i csn spawn a PR on Monday with an example then once you are happy with the variant wording i can propogate it to the other boards.

dlabun, 2017-10-20:
Awesome. I am also thinking at this point we should probably start a changelog.

micooke, 2017-10-21:
Sounds like a plan Stan.

There have been quite a few bug fixes & enhancements since the last release so good to start capturing this. 

Im also looking in to #176 - the arduino core lib is quite different, i want to see if that will resolve the issue.

dlabun, 2017-10-21:
Something to keep in mind, we are using an older version of Nordic SDK... Maybe we're bumping up against limitations or bugs in the older SDK.

netkruzer, 2018-01-04:
I put "#define CONFIG_NFCT_PINS_AS_GPIOS" in my INO file an I can't seem to get pins p0.09 and p0.10 to work as inputs.

I'm using Generic nRF52.

micooke, 2018-01-04:
The SDK gets compiled separately and linked in, so you need that ```#define``` at compile time. If you check out PR #208, there is a board configuration option that does this for the generic NRF52

netkruzer, 2018-01-04:
I know this might be a dumb question, but how do I go about activating this on my build? Do I modify the boards.txt file or something else?

micooke, 2018-01-04:
Yep, you could just append ```-DCONFIG_NFCT_PINS_AS_GPIOS``` to build.extra_flags on line 42.

Note that if you are using the soft device, you will need to append it to line 54.

Or you could just pull the changes i made in #208 :wink:

netkruzer, 2018-01-05:
Thanks! That worked perfect.

`Generic_nRF52832.build.extra_flags=-DNRF52 -DCONFIG_NFCT_PINS_AS_GPIOS`
...
`Generic_nRF52832.menu.softdevice.s132.build.extra_flags=-DNRF52 -DS132 -DNRF51_S132  -DCONFIG_NFCT_PINS_AS_GPIOS`

Really appreciate the help!

dlabun, 2018-09-03:
Closing due to lack of activity

closed 2018-09-03 by dlabun

micooke, 2018-09-03:
Thanks @dlabun, i should have closed this earlier.

In response to https://github.com/sandeepmistry/arduino-nRF5/pull/208#issuecomment-414126935
These additions were moved to the Taida Century variant in my smartwatch repo https://github.com/micooke/arduino-nRF5-smartwatches/blob/master/boards.txt

Protonerd, 2019-04-02:
Hi guys, sorry for necro-posting, I'm having the same problem, using an Adafruit Feather BLE. I edited the boards.txt under ..\Adafruit_nRF52_Arduino-master to add under the
#**********************************************
# Feather nRF52832 Board
#**********************************************
...
# Build
...
feather52832.build.extra_flags=-DNRF52832_XXAA -DNRF52 -DCONFIG_NFCT_PINS_AS_GPIOS

But it still refuses to toggle the 2 NFC pins as GPIOs. What am I doing wrong?

Protonerd, 2019-04-02:
xxx

micooke, 2019-04-03:
As a guess, you are selecting a board with the adafruit core not the nRF5 core from sandeeps repo. Ask Adafruit, or use the NRF52832 Generic in this core