sandeepmistry/arduino-nRF5 #210: Persistent Storage

gpgabriel, 2017-11-17:
Hi,

I might be wrong but from the current source code of this repository I think there's no way to use FDS or pstorage as they're not included from the SDK. (https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk5.v11.0.0%2Flib_pstorage.html)

Is there any plan to add this functionality?

Gabriel

dlabun, 2017-11-21:
Correct, PStorage is not included. I do not believe there are any plans currently to add it as it would require relying more on the Nordic SDK.

sandeepmistry, 2017-11-21:
@gpgabriel checkout's @d00616's library: https://github.com/d00616/arduino-NVM

closed 2017-11-21 by sandeepmistry

DhirajGehlot, 2017-12-31:
Hello @sandeepmistry 

Correct me if I am wrong. According to this link: http://infocenter.nordicsemi.com/index.jsp?topic=/com.nordic.infocenter.s130.sds/dita/softdevices/s130/sd_resource_reqs/hw_block_interrupt_vector.html&cp=3_7_2_0_6_0
if softdevice is enabled, the flash can be written only through SoftDevice API sd_flash_write commands.
https://github.com/d00616/arduino-NVM : This uses writing to flash using registers.