sandeepmistry/arduino-nRF5 #213: Custom I2C pins for generic variants

honnet, 2017-11-21:
Hi again @sandeepmistry & @dlabun ;)

Cheap [nRF52 modules](https://www.aliexpress.com/item/wholesale-Holyiot-TinyBLE-nRF52832-Bluetooth-low-energy-module-BLE-5-0-for-Bluetooth-mesh/32840369737.html) are getting common, so this [Generic variant](https://github.com/sandeepmistry/arduino-nRF5/blob/master/variants/Generic/) that you offer here is really a great feature.

Currently, the the default Wire object is initialized with default I2C pins that are forced:
- [Wire_nRF52.cpp#L387](https://github.com/sandeepmistry/arduino-nRF5/blob/master/libraries/Wire/Wire_nRF52.cpp#L387)
- [Generic/variant.h#L96-L100](https://github.com/sandeepmistry/arduino-nRF5/blob/master/variants/Generic/variant.h#L96-L100)

It makes sense for most boards, but maybe not for the generic variants as these pins could be anywhere.

It would be a pleasure to implement + PR an improvement allowing more flexibility but it seems appropriate to hear your thoughts about the following possible solutions:

1. Modifying a bit the declaration/implementation of `begin()` for [master & slave](https://github.com/sandeepmistry/arduino-nRF5/blob/master/libraries/Wire/Wire.h#L42-L44) to allow setting I2C pins with optional default values (as in [here](http://en.cppreference.com/w/cpp/language/default_arguments)) might be the most appropriate.
2. Using `#define` instead of `const` [here](https://github.com/sandeepmistry/arduino-nRF5/blob/master/variants/Generic/variant.h#L99-L100), should allow using `#undef` but it might not be very reliable, so simply removing the [default values](https://github.com/sandeepmistry/arduino-nRF5/blob/master/variants/Generic/variant.h#L70-L100) seems more convenient. It would also take care of [SPI](https://github.com/sandeepmistry/arduino-nRF5/blob/master/variants/Generic/variant.h#L80-L89) and [UART](https://github.com/sandeepmistry/arduino-nRF5/blob/master/variants/Generic/variant.h#L74-L75), and users would just have to declare/define the pin constants/macros in their personal files).

Any thought?
Cedric ;)

dlabun, 2017-11-21:
The generic device is really more of a template, if you're building a custom device you really should be taking the generic device and modifying it to your needs. Of course, that's not to say we can't make the process more robust.

honnet, 2017-11-21:
Happy to read that you're open to make this generic variant more robust :D
Would you have a preference if I implement it and do a pull request? (solution 1, 2 or something else?)

A 3rd simpler solution could be to add a `#define` around the "[default forced values](https://github.com/sandeepmistry/arduino-nRF5/blob/master/variants/Generic/variant.h#L70-L100)" to allow disabling them.
Does that sound better?

Thanks again ;)

sandeepmistry, 2017-11-21:
@honnet 

> A 3rd simpler solution could be to add a #define around the "default forced values" to allow disabling them.
> Does that sound better?

I like that! If you can that to work, please submit a pull request.

honnet, 2017-11-24:
Hi again @sandeepmistry & @dlabun !
Here is a 1st attempt: https://github.com/honnet/arduino-nRF5/commit/fa4eea4bc0e8116bb482f58db61d744546dbb700
It simply adds an option (in the Arduino board option menu) that can disable the forced pinout from the template.
Let me know how it can be improved!
Cedric ;)

honnet, 2017-12-11:
Hey, @sandeepmistry & @dlabun ;)
Is there anything that could be done to improve this proposition?
honnet/arduino-nRF5@fa4eea4
Thanks!
Cedric ;)

sandeepmistry, 2017-12-18:
@honnet would it be possible to do something like:

start of sketch
```
#define PIN_WIRE_SDA 6
#define PIN_WIRE_SCL 7
#include <Arduino.h>
```

variant:
```
#ifndef PIN_WIRE_SDA
#define PIN_WIRE_SDA 20
#endif

#ifndef PIN_WIRE_SCL
#define PIN_WIRE_SCL 21
#endif
```

Otherwise, maybe we can add a `Wire.setPins(sda, scl)` API that you call before `Wire.begin(...)`?

honnet, 2017-12-19:
Sadly the .ino sketch seems to be compiled at the end so the `#ifndef` solution would apparently not work as is. (if arduino used a Makefile I would know what to do but their tool chain is still a bit mysterious to me)

The `Wire.setPins(sda, scl)` solution is nice, I could do it for the UART and SPI too...
...and I'll probably do another commit to allow using the other Analog inputs ;)

Sounds good ?

sandeepmistry, 2018-01-14:
Closed via PR #221 :)

closed 2018-01-14 by sandeepmistry