sandeepmistry/arduino-nRF5 #215: error uploading softdevice

cheeta1, 2017-12-10:
i get this output when i try to upload softdevice in nrf52832

Open On-Chip Debugger 0.10.0-dev-00254-g696fc0a (2016-04-10-10:13)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
0x4000
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 10000 kHz
Info : Unable to match requested speed 10000 kHz, using 4000 kHz
Info : Unable to match requested speed 10000 kHz, using 4000 kHz
Info : clock speed 4000 kHz
Info : STLINK v2 JTAG v17 API v2 SWIM v4 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.237541
Info : nrf52.cpu: hardware has 0 breakpoints, 2 watchpoints
Info : Halt timed out, wake up GDB.
Error: timed out while waiting for target halted

Error while flashing SoftDevice.

dlabun, 2017-12-10:
From the looks of the error message the chip is not responding. Are you sure you are connected properly?

cheeta1, 2017-12-11:
hi @dlabun the board is connected correctly. i am using stlink V-2 to program. there is some problem that is mentioned in following lines

Info : nrf52.cpu: hardware has 0 breakpoints, 2 watchpoints
Info : Halt timed out, wake up GDB.
Error: timed out while waiting for target halted

it says that it detects nrf52 but something about not halting is making errors. i have read on nordic form that if the chip is protected then this kind of error may occur. here is the link

https://devzone.nordicsemi.com/question/78890/programming-nrf52-with-openocd/

dlabun, 2017-12-11:
Which board are you using and have you ever programmed it before? It does sound like the chip is protected.

cheeta1, 2017-12-11:
i am using skb369 module. i have programmed them many times. these modules i got this week are showing this error.

dlabun, 2017-12-11:
Just to rule it out, try programming a different board. It's possible new modules you bought are protected or just defective.

cheeta1, 2017-12-11:
i think these are protected. they have a default firmware working on it. can you give me any idea how to remove protection so that i can reprogram them?

dlabun, 2017-12-12:
That's really a question for Nordic or whomever makes those modules... I think the Nordic programming tools may be able to completely wipe them.

alextsam, 2017-12-13:
You can use the openocd that is bundled with arduino-nrf5 to unlock the nrf52. The following worked for me (different nrf52 modules but same problem):
```
openocd.exe -d2 -f interface/stlink-v2.cfg -c "transport select hla_swd; set WORKAREASIZE 0x4000;" -f target/nrf52.cfg -c init -c "reset init" -c halt -c "nrf51 mass_erase; reset; shutdown;"
```

dlabun, 2017-12-25:
Close due to lack of activity

closed 2017-12-25 by dlabun