sandeepmistry/arduino-nRF5 #216: Using newer softdevice versions

ayavilevich, 2017-12-16:
Softdevice 132 is now at version 5.0.0 but the package only knows to flash 2.0.1 .

What will it take to use newer/different versions of softdevices with this package?
If I flash a different version myself will it work ok with the code generated?

dlabun, 2017-12-17:
At the time this core was developed the Nordic SDK only supported 2.0.1. You can certainly try flashing a new version but it really depends on what chip you are using.

dlabun, 2018-01-06:
Closing due to lack of activity

closed 2018-01-06 by dlabun

bobr-rigado, 2019-01-09:
Are there plans to update to the newer SoftDevices, especially since they will be required in order to obtain an end-product declaration with the Bluetooth SIG?