sandeepmistry/arduino-nRF5 #217: How can i use this library with arduino primo core?

ttn115, 2017-12-25:
Hi,

I develop a program using Arduino Primo, and I want to layout a new PCB board with new nRF52832 chip. The new chip may not have any bootloader. So I think this library is a solution?

The Arduino Primo Core have SWD interface and has a bootloader,  so I link the core by using segger trying to flash the core as picture show
[image]
I try to follow README.md on the step "Flashing a SoftDevice", i not sure which board i should choose?
Any suggestion or advice?

jeremypoulter, 2017-12-25:
I found that I have to flash the Softdevice with Arduino IDE first

jeremypoulter, 2017-12-25:
Ignore that I was thinking about platform IO... And hit send by mistake

dlabun, 2017-12-25:
The Primo Core is not currently supported by this repository so you are not going to be able to program it without modifying the code. 

As for your custom board, are you using the same schematic as a support board or is your schematic customized?

ttn115, 2017-12-26:
Thanks for your reply.
Base on i don't have much experience, i may ask some basic question.
@jeremypoulter 
How do you flash your arduino primo or core?
And how do you restitution them?
After i flash the board to blank, how can i burn our code into the chip?
@dlabun 
I also have arduino primo. Does it support by this repository? 
Yes, our schematic will be customized.
Any advice would be appreciate.

dlabun, 2017-12-26:
The Primo and Primo Core are official Arduino products so they're already supported in the Arduino IDE. You only need to follow the Primo set up directions to be able to work with those boards.

ttn115, 2017-12-27:
@dlabun 
Thanks for reply.
Now i think the arduino is easy for POC, but some arduino is not suitable for people create a now product. 
Or maybe that just not suitable for us now. I'll keep learning to solve my problem.

dlabun, 2017-12-27:
My recommendation is to simply take the Primo Core and just customize the board design to meet your needs. That way you don't need to write a custom board definition with this repository and can keep using the official Arduino core.

dlabun, 2017-12-27:
I also highly recommend checking how you are connecting the J-Link to your Primo, it looks to me like you might have your adapter board connected backwards.

ttn115, 2017-12-28:
@dlabun Thanks for reply.
Yes, i'm still trying to connect the J-Link to your Primo, and confuse about bootloader、softdevice、my code. Wonder to know the relationship in the primo. Is that any different from nrf52 DK?
And i decide to learn form mass production of nrf52 then back to develop my code, hope that will be more effiency then what i do before.

dlabun, 2017-12-28:
The Primo is very much different from the nRF52 DK and the boards supported by this repository. As far as I know the Primo does not have a bootloader on it, which is why you need to use an external programmer. If you need more help working with the Primo you should contact Arduino since it's an official board.

closed 2018-01-06 by dlabun