sandeepmistry/arduino-nRF5 #220: millis() does not work as it should it overflows after ~8 minutes (overflows never gets incremented)

TraxXavier, 2018-01-06:
Hello,

I have a problem with the millis() function, it always overflows after roughly 8 minutes of operation, also the overflow handler for the RTC1 (RTC1_IRQHandler) is never invoked hence the overflows variable never incremented.

Also as far as I see the delay() function can fail if an overflow of millis offurs.

TraxXavier, 2018-01-07:
the fix:
delay.c

uint32_t millis( void )
{
  uint64_t ticks = (uint64_t)((uint64_t)overflows << (uint64_t)24) | (uint64_t)(NRF_RTC1->COUNTER); // Trax Fix: 24 bit counter not 32

  return (ticks * 1000) / 32768;
}

uint32_t micros( void )
{
  uint64_t ticks = (uint64_t)((uint64_t)overflows << (uint64_t)24) | (uint64_t)(NRF_RTC1->COUNTER); // Trax Fix: 24 bit counter not 32

  return (ticks * 1000000) / 32768;
}

wiring.c

void init( void )
{
  NVIC_SetPriority(RTC1_IRQn, 15);
  NVIC_ClearPendingIRQ(RTC1_IRQn);
  NVIC_EnableIRQ(RTC1_IRQn);

  NRF_CLOCK->LFCLKSRC = (uint32_t)((CLOCK_LFCLKSRC_SRC_RC << CLOCK_LFCLKSRC_SRC_Pos) & CLOCK_LFCLKSRC_SRC_Msk);
  NRF_CLOCK->TASKS_LFCLKSTART = 1UL;

  NRF_RTC1->PRESCALER = 0;
  NRF_RTC1->INTENSET = RTC_INTENSET_OVRFLW_Msk; // Set the overflow interrupt properly
  NRF_RTC1->TASKS_START = 1;
}

sandeepmistry, 2018-01-07:
@TraxXavier could you please open a pull request for this suggestion.

TraxXavier, 2018-01-09:
will have to look into how to do that :D

sandeepmistry, 2018-02-26:
@TraxXavier any progress on this?

jeremypoulter, 2018-02-28:
I think this has already been fixed in #87

sandeepmistry, 2018-03-11:
@jeremypoulter thanks, let's close for now then.

closed 2018-03-11 by sandeepmistry