sandeepmistry/arduino-nRF5 #222: Writing data to Flash using Pstorage

DhirajGehlot, 2018-01-11:
Hello Everyone,

This is continuation of issue Persistent Storage #210 
Correct me if I am wrong. According to this link: http://infocenter.nordicsemi.com/index.jsp?topic=/com.nordic.infocenter.s130.sds/dita/softdevices/s130/sd_resource_reqs/hw_block_interrupt_vector.html&cp=3_7_2_0_6_0
if softdevice is enabled, the flash can be written only through SoftDevice API sd_flash_write commands.
https://github.com/d00616/arduino-NVM : This uses writing to flash using registers.

dlabun, 2018-01-13:
You are correct, if you are using a SoftDevice you can only access the flash via the API.

sandeepmistry, 2018-01-14:
There's nothing in this repo that writes to flash, I suggest you open an issue here to discussion this with the arduino-NVM maintainers: https://github.com/d00616/arduino-NVM/issues

closed 2018-01-14 by sandeepmistry