sandeepmistry/arduino-nRF5 #223: waveshare ble400 clone+nrf51822 on arduino using usb only

iotlearner0level, 2018-01-12:
Sorry if this question has been asked before but i want to know whether it is possible to program the device on arduino using either stlink clone or usb only (ie only one of these, after flashing the s130 firmware)?

right now, with instructions on github page, i use swd+sclk pins on stlink clone as well as usb to power+RX+TX on arduino otherwise it says unable to find port...BLEperfipheral library works fine.

I am new to both nrf51 as well as ARM so maybe i am missing some information.

dlabun, 2018-01-12:
The only way to program via USB only would be to write your own bootloader. I haven't seen any bootloaders for the nRF series just yet.

iotlearner0level, 2018-01-13:
is it possible using only stlink (without usb connection)?

dlabun, 2018-01-13:
Yes, the ST Link is supported but there's no guarantees about a clone.

iotlearner0level, 2018-01-13:
i've a clone and it works with stlink drivers. i even upgraded the stlink firmware on the clone from the stlink website. unfortunately i also need to plugin the usb to waveshare ble400 otherwise it doesn't work. it says can't find the port.

My question is how to get it working using only SWDIO/SWDCLK without the usb cable?

antother question (could not find an answer) if i flash the firmware s130 using arduino>tools>burn bootloader, is it possible to make OTA updates as in the case of wifi SoC like esp8266 using Arduino OTA?

dlabun, 2018-01-13:
In order to program the board it has to have power... If you don't want to use the USB cable then you will need to connect an external power supply.

iotlearner0level, 2018-01-13:
i get an error in arduino can't open port or device not connected if usb (ie usb to serial) is not connected. i think i can power it using the stlink clone 3.3V and GND

dlabun, 2018-01-13:
You would have to check if your programmer is capable of powering the board, Some programmers only supply a few milliamps while others won't supply any power.

iotlearner0level, 2018-01-14:
ok, looks like the programmer is now "bricked" it says can't connect to device, earlier it was ok. that could be bacause i connected oscilloscope ground to its ground (ground loop?)

earlier i managed the adafruit ssd 1306 128x64 i2c oled demo, works ok but a bit slowly.

now that the programmer seems to be bricked (don't know much about ARM world), is there any possibility of arduino over the air updates via ble? i read something about bluetooth DFU mode...so if i generate a firmware in arduino, it should be possible to flash it?

how do i check if there is a hardware failure in the stlinkv2 clone?

sandeepmistry, 2018-01-14:
Closing this for now, as the original topic is answered. I suggest you contact the programmer manufacturer and or board provider as to why it's "bricked".

You should select the relevant programmer from "Tools -> Programmer" and then use "Sketch -> Upload Using Programmer" to flash the sketch without a serial port present.

DFU requires a custom boot loader and is not supported by this project currently.

closed 2018-01-14 by sandeepmistry