sandeepmistry/arduino-nRF5 #227: New release?

sandeepmistry, 2018-01-14:
@dlabun any thoughts on if we should have a new 0.5.0 release of the core soon?

The only outstanding items I think we should consider waiting for are:

 * PR #224 and #226 
 * A PR from @bojh to enable the reset pin on a few variants (https://github.com/sandeepmistry/arduino-nRF5/pull/91#issuecomment-355980886)

dlabun, 2018-01-14:
I think a new release is a good idea. What about PR #214?

sandeepmistry, 2018-01-19:
I've merged #214, and you've merged #224  and #226 :)

PR #229 is still pending.

Do you have time to make the release? Otherwise, I can schedule some time to do so.

sandeepmistry, 2018-01-29:
PR #229 is merged now.

dlabun, 2018-02-05:
I am going to target Saturday morning (02/10/18) for the new release, short of any major issues arising.

sandeepmistry, 2018-02-10:
@dlabun awesome!

I noticed you made a tag already for this. This might be already on your list, but we'll need to update https://github.com/sandeepmistry/arduino-nRF5/blob/master/platform.txt#L19 and re-tag.

What do you think about starting a Github wiki page on a checklist for a making a release? At a later date we can start automating the check list with some build scripts + CI.

dlabun, 2018-02-10:
I pulled the new tag last night as a reminder for my thick head to work on this today.

Definitely like the sound of automating the release process... It took a few attempts for me last time.

dlabun, 2018-02-13:
0.5.0 is live and Travis is passing.

closed 2018-02-13 by dlabun

sandeepmistry, 2018-02-16:
@dlabun awesome, thanks for handling the release!

How did the process go this time? Regarding automation system for releasing in the future, some possibilities include:

1) Having some cross platform scripts to do it from a maintainers PC
2) Making a tag and letting Travis CI handle the release

I think one manually step will be to include the boards supported in the JSON file, although maybe this can be pulled from boards.txt

Any more options you can think of? Any preferences on them?

dlabun, 2018-02-16:
It went fine this time around since I actually knew what I was doing. Adding the boards in is a pain as they're not in alphabetically order in boards.txt.

sandeepmistry, 2018-02-22:
Here's a rough Python script to make add a new release tag to the JSON, would this help?

```python
import fileinput
import json
import md5
import requests
import sys

packageJsonFile = 'docs/package_nRF5_boards_index.json'

version = sys.argv[1]
url = 'https://github.com/sandeepmistry/arduino-nRF5/archive/' + version + '.tar.gz'
archiveFileName = 'arduino-nRF52-' + version + '.tar.gz'

# download url
archive = requests.get(url)
archiveSize = len(archive.content)
archiveMd5 = md5.new()
archiveMd5.update(archive.content)

# read board names
boards = []
with open('boards.txt', 'r') as input:
	for line in input:
		if ".name=" in line:
			boards.append( line.split('=')[1].strip() )

boards.sort()

# parse existing package json
package = json.load(open(packageJsonFile))
platforms = package['packages'][0]['platforms']

# copy last platform entry
platform = platforms[-1]

# update fields
platform['version'] = version
platform['url'] = url
platform['archiveFileName'] = archiveFileName
platform['checksum'] = 'MD5:' + archiveMd5.hexdigest()
platform['size'] = str(archiveSize)

# fill in boards section
platform['boards'] = []
for board in boards:
	platform['boards'].append( dict(name=board) )

# add new platform version
package['packages'][0]['platforms'].append(platform)

# output file
with open(packageJsonFile, 'w') as outfile:
	json.dump(package, outfile, indent = 2, ensure_ascii = False)
```