sandeepmistry/arduino-nRF5 #228: Error while flashing SoftDevice - J-Link EDU

Gius-8, 2018-01-16:
Hi guys, 
I'm trying to load the arduino core on a generic nRF51822 but I always get the same error:
```
C:\Users\gius_\AppData\Local\Arduino15\packages\sandeepmistry\tools\openocd\0.10.0-dev.nrf5/bin/openocd.exe -d2 -f interface/jlink.cfg -c transport select swd; set WORKAREASIZE 0; -f target/nrf51.cfg -c init; halt; nrf51 mass_erase; program {{C:\Users\gius_\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.4.0/cores/nRF5/SDK/components/softdevice/s110/hex/s110_nrf51_8.0.0_softdevice.hex}} verify reset; shutdown; 
Open On-Chip Debugger 0.10.0-dev-00254-g696fc0a (2016-04-10-10:13)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
0
cortex_m reset_config sysresetreq
adapter speed: 1000 kHz
jaylink: Failed to open device: LIBUSB_ERROR_NOT_SUPPORTED.
jaylink: Failed to open device: LIBUSB_ERROR_NOT_FOUND.
Info : No device selected, using first device.
Error: No J-Link device found.

Error while flashing SoftDevice.
```
I use the latest j-link drivers (JLink_V622f). I tried with Arduino 1.8.5 and 1.6.13.
Can someone help me in some way?
Thanks so much

dlabun, 2018-01-16:
Did you follow the directions on the readme file to setup the JLINK?

Gius-8, 2018-01-16:
Hi @dlabun , 
yes, I followed the README:
https://github.com/sandeepmistry/arduino-nRF5#driver-setup-for-segger-j-link

Through Zadig I need to install libusb or WinUSB?
Do you think I should remove the J-Link driver?

Thanks :)

Gius-8, 2018-01-18:
Hi, I've solved by uninstalling and reinstalling the j-link drivers. Then I used zadig.
Thank you

closed 2018-01-18 by Gius-8

netafimamarel, 2019-07-03:
@Gius-8 which one you select on zadig ?

JosefGst, 2020-08-13:
for me it worked with WinUSB