sandeepmistry/arduino-nRF5 #231: hex file uploading using nrfgo studio

cheeta1, 2018-01-24:
hi there i have compiled a code on nrf51 and tried to upload the soft device and hex file generated by the arduino using nrfgo studio and it give following error. any ideas  how to solve this issue?
[image]

cheeta1, 2018-01-24:
@dlabun  did you reach to any conclusion on this issue?  
https://github.com/sandeepmistry/arduino-nRF5/issues/161

cheeta1, 2018-01-24:
there is some address conflict the arduio application hex file starts with following

:020000021000EC
:10B0000000400020D5DE010015DF010017DF010040
:10B010000000000000000000000000000000000030
:10B0200000000000000000000000000019DF010027
:10B0300000000000000000001BDF01001DDF010018
:10B040001FDF01001FDF010021E401001FDF0100FD
:10B050001FDF0100000000001FDF01001FDF0100F3
:10B060001FDF01001FDF01001FDF01001FDF0100E4
:10B070001FDF01001FDF01001FDF01001FDF0100D4
:10B080001FDF01006DDF01001FDF01001FDF010076
:10B090001FDF01001FDF01001FDF01001FDF0100B4
:10B0A0001FDF01001FDF01000000000000000000A2

and the softdevice ends with following

:10B5D00017A00100679F010083A00100DDA001000A
:10B5E000EDA0010019A1010093A60100DBA6010056
:10B5F00011A7010043A701006BA7010093A7010059
:10B60000D5A70100F5A701000DA801004DA8010074
:10B610009F120100A7120100B512010047A101000D
:10B6200061A1010035A101003FA101006DA1010050
:10B63000A3A10100EFA10100FDA101000BA20100E7
:10B6400017A2010025A2010033A201003FA20100C0
:10B65000000000004B8F0000A18F0000B78F00009A
:10B66000ADA80100559C01001B9D01000FAC01001D
:10B670003DAC010075AC0100951001006D15010095
:10B6800000100200A4B6010008000020A001000084
:10B6900044110000C8B60100A8010020E00F00001E
:10B6A0008011000001593601000100683720FB3489
:10B6B0009B5F80041F800010022001337F0102E4A1
:08B6C00029E4D1AF01000000F4
:00000001FF

so there is address conflict i also tried the mergehex tool of nordic and it gives following error

[image]

dlabun, 2018-01-25:
What happens if you flash your sketch from the Arduino IDE? It's difficult to say if this is problem with out the hex file is being generated or if it's the Nordic tool.

cheeta1, 2018-01-25:
if i flash directly with Arduino ide then there is no issue. i have tried arduino primo files and they are uploaded using nrfgo studio successfully. it just that the file created by this repository does not either merge with mergehex tool or uploaded by nrfgo studio

you can clearly see that the  soft device ends with this address :10B6B and application file starts with :10B01 sor there is a conflict between :10B01-:10B6B addresses. 

any idea how we can solve this?

cheeta1, 2018-01-25:
one thing i have noticed that arduino primo generate hex file starting from :10C00 as give below

:020000021000EC
:10C00000000001205DF801009DF801009FF801008B
:10C01000A1F80100A3F80100A5F80100000000004C
:10C02000000000000000000000000000A7F8010070

thats why their files can be merged and uploaded easily through nrfgo studios.

can we also change the starting address of hex file created by this library to :10C00? if yes then do tell me where to look.

thanks again for your co-operation.

cheeta1, 2018-01-25:
there was some issue with the soft device i downloaded new one and it works fine. 

thanks you for your support

closed 2018-01-25 by cheeta1