sandeepmistry/arduino-nRF5 #234: PlatformIO as alternative build system for Arduino

ivankravets, 2018-02-05:
Hi,

I would like to provide a PR with information and build script for @PlatformIO. I did the same for Arduino Core for ESP8266 and ESP32:

- https://github.com/esp8266/Arduino#using-platformio
- https://github.com/esp8266/Arduino/blob/master/tools/platformio-build.py
- https://github.com/espressif/arduino-esp32#installation-instructions
- https://github.com/espressif/arduino-esp32/blob/master/tools/platformio-build.py

PlatformIO users already use your Arduino Core for nRF51/52. The build script is located here https://github.com/platformio/builder-framework-arduino-nrf5

Also, I would like to extend your CI script and add support for PlatformIO to ensure that examples work with PlatformIO too.

All things including debugging work in the "1-click" with PlatformIO and 10 different IDEs.

Thanks in advance!

dlabun, 2018-02-05:
This repository is really intended to target the Arduino IDE... If there's is already a build script available for PlatformIO I am not sure it's necessary for us to make additional changes. I honestly don't know anything about PlatformIO (let alone use it) so supporting it here officially would be difficult.

ivankravets, 2018-02-05:
If you will have a problem when something works with Arduino IDE and does not with PlatformIO, you can ping me and our team will help ASAP.

Nevertheless, could we place some information to README that people also can use PlatformIO?

dlabun, 2018-02-05:
In my opinion I would say no, PlatformIO is outside the scope of this repository. Since Sandeep is the owner I will defer to his decision on the matter.

sandeepmistry, 2018-02-10:
So I totally get why PlatformIO would like this added to the repo, but it feels like a one way street.

How would the maintainers of this project/repo benefit from these additions?

jeremypoulter, 2018-02-10:
You will get some extra visibility, the only reason I am using this project
is because it is supported in Platform IO.

ivankravets, 2018-02-10:
This PR brings benefits for the end users. They can still use Arduino IDE or PlatformIO IDE if they need a more advanced development environment with 1-click debugging, etc. What is more, we provide integration with [10 other IDEs](http://docs.platformio.org/en/latest/ide.html), including cloud and standalone.

You did a great work and community supported you (>200 stars). If this repo has been created for community, we could bring more users.

If you don't want to have Python/PlatformIO script within this repo, we can add only information to README with a short information how to use this repo with PlatformIO.

dlabun, 2018-02-10:
I'd still like to know why we need to maintain the Python scripts on our end... If PlatformIO already maintains them, why should the burden be shifted onto us? To me that would mean ensuring compatibility with two separate IDE / build environments.

ivankravets, 2018-02-10:
@dlabun please read my last paragraph. If you don't want to test this source code (CI) with PlatformIO Build System, we will not add it to PR. Just only information README how to use this repo with PIO.

dlabun, 2018-02-10:
That still places responsibility on us to maintain those instructions... If something changes on the PlatformIO side we really wouldn't have any way to know about it until we start getting reports that the instructions are broken.

ivankravets, 2018-02-10:
We don't modify your source code. We use the same which pointed to release tag. The build flags are the same, we maintain them manually.

Nevertheless, we've got your position. Let's wait for @sandeepmistry.

dlabun, 2018-02-10:
I'm strictly talking about the ReadMe file and keeping those instructions up to date... Plus the ESP8266 example you cite reads like advertisement, there's no actual install instructions. When we received a PR to add support for the Black Magic Probe last year it was literally just instructions on how to setup the drivers and nothing more.

ivankravets, 2018-02-10:
> Plus the ESP8266 example you cite reads like advertisement

Have you tried PlatformIO? You do not need to do extra steps with PlatformIO. Just install our IDE, for example, an extension for VSCode, create new project for nRF board and press upload button. 

We have an opposite position to Arduino.cc. We allow any user, maker, vendor, company to add own platform, board, a framework to our registry. In a case with Arduino.cc, you need to $PAY to be listed by default in "Board Manager". That is why people write these manuals with dozen lines how to install "yet another board" to Arduino IDE.

Please don't compare Arduino vs PlatformIO, these organizations could not be comparable. We found PlatformIO for Community but not for "just make a money". We love and hear our Users.

I respect Arduino Community and people/developers who contribute. For me, Arduino = community/world but not to CC.

* Our philosophy - http://docs.platformio.org/en/latest/what-is-platformio.html#problematic
* True story - https://arduinohistory.github.io

dlabun, 2018-02-10:
Umm, not sure what any of that has to do with the fact that the "instructions" you cite are not instructions. You have been talking around every question asked so far. I haven't actually seen anyone compare Arduino to PlatformIO so far, it's only been a discussion on the value of this. 

You should also be careful about throwing rocks at Arduino... A link from your own website: [http://platformio.org/pricing](url)

And yes, I did try PlatformIO before joining this project but it didn't support any of the chips I was working with.

ivankravets, 2018-02-11:
> Umm, not sure what any of that has to do with the fact that the "instructions" you cite are not instructions. 

We would like to add info similar to https://github.com/esp8266/Arduino#using-platformio

Do you have any objectives?

> You should also be careful about throwing rocks at Arduino...

Please don't play with words... I edited comment above and changed Arduino.cc to Arduino LLC for better reading... 

>  A link from your own website: http://platformio.org/pricing

What is wrong with that link?

> And yes, I did try PlatformIO before joining this project but it didn't support any of the chips I was working with.

Which chips are missed in our registry?

dlabun, 2018-02-11:
Again, That's an advertisement and not instructions on how to use PlatformIO. Instructions take the form of "Step 1: Download driver XYZ, Step 2: Execute script ABC", etc, etc.

As @sandeepmistry asked, How would the maintainers of this project/repo benefit from these additions?  I highly recommend answering that before this thread is closed.

kzyapkov, 2018-02-12:
The biggest gain out of this, for both end-users and maintainers, would be CI testing against platformio. It does bring advantages over Arduino in many scenarios.

The rest is politics, which can be enjoyable to watch.

ivankravets, 2018-02-12:
> Again, That's an advertisement and not instructions on how to use PlatformIO

O, I think that is a minimal problem which we have in this discussion. We can prepare PR which you will like to see. Steps, drivers?, ABC scripts? 👀 , etc.

> How would the maintainers of this project/repo benefit from these additions? I highly recommend answering that before this thread is closed.

I even don't know what should I answer. Please read comments above from community, they described all benefits. If you mean "money", we don't make money on Arduino users. All our features are free in Community plan. We propose MORE and MORE for FREE in comparison with Arduino IDE.

The only a few things related to advanced development are paid. They are popular for enterprise customers or advanced developers who in 99% don't use Arduino Framework at all. We support 10+ different frameworks and enterprise customers have own development platforms based on our build system.

What is more, ~90% of Arduino users haven't heard about embedded debugging before or, better, don't have ideas how to use it. Only a few Arduino boards have debugging interface or onboard debug probe. The same with local and remote unit testing.

Finally, from my point of view, the main aim of this PR is to attract more users for your Arduino Core within our ecosystem, and from our side, allow users to try PlatformIO who have not heard about it before. Again, we will not force them to use PlatformIO. 

I understand that you work for Arduino LLC and this PR looks strange for you. Also, we like [this post](https://www.instagram.com/p/BeGOLkaAJMY/?taken-by=mbanzi) by @mbanzi. However, this is open source, community will decide which software to use. 

P.S: In PlatformIO, we respect Arduino community and even Arduino IDE. Developers who work for Arduino Cores and IDE prepare new customers for our ecosystem. We know that in the current state we are not good for children or beginners. But we are very happy to be a continuation for Arduino IDE. We even have special button "Import Arduino Project" which allow users easily migrate to PlatformIO.

Regards,
- Dr. Ivan Kravets
- Founder and CEO at PlatformIO

dlabun, 2018-02-12:
> I understand that you work for Arduino LLC and this PR looks strange for you. 

Does anybody know the phone number for Human Resources at Arduino? I want to know why I haven't been getting paid.... or where my office is.

There's a saying I've heard many times "Arguing with an engineer is alot like wrestling in the mud with a pig. After a couple of hours you realize the pig likes it." Using irrelevant and unsubstantiated facts are by far one of my most favorite marketing tactics I see at work and I get the biggest laugh from them.

FYI: this isn't a PR... It's just an issue ticket. You haven't actually submitted a PR with any proposed changes. So far you're doing nothing but getting muddy.

ivankravets, 2018-02-12:
> Does anybody know the phone number for Human Resources at Arduino? I want to know why I haven't been getting paid.... or where my office is.

Why did you decide that I answered you? 👀
You asked me to answer on "How would the maintainers of this project/repo benefit from these additions?". 

> Using irrelevant and unsubstantiated facts are by far one of my most favorite marketing tactics I see at work and I get the biggest laugh from them.

Maybe, because you don't like to hear [truth](https://arduinohistory.github.io)?

> FYI: this isn't a PR... It's just an issue ticket.

Thanks, God! Now I understand a difference between an issue and pull request. I'm newbie here 🐠 

P.S: https://github.com/sandeepmistry/arduino-nRF5/issues/234#issue-294464241 ... "I would like to provide a PR"... I don't want to spend time on PR which will be refused. I visited [Contributors](https://github.com/sandeepmistry/arduino-nRF5/graphs/contributors) list and realized that first need to ask before working on it. If you read me carefully, I had an idea with PR which will contain PlatformIO build script and integration to CI. Normally, developers should discuss these changes. That is why we here.

Indeed, I would like to hear @sandeepmistry. @dlabun, I've said [ABOVE](https://github.com/sandeepmistry/arduino-nRF5/issues/234#issuecomment-364680292), that I got you! Thanks.

dlabun, 2018-02-12:
No, sadly you don't understand that I don't work for Arduino... I'm nothing but an Arduino end user who's had to put up with the same CC vs ORG crap as everyone else. Quite honestly, I could careless about the Ardiuno "truth" as it really doesn't affect the fact that I can still buy a Zero from my suppliers.

radek-senfeld, 2018-02-12:
PlatformIO support makes sense as it's a great tool for advanced users. I can do some maintenance if needed.

curtpw, 2018-02-12:
One of the reasons I use this project is because it is in no way tied to a for-profit business. "Advanced users" are going to be using the Nordic SDK directly. Arduino is great for scientific research because it is accessible to anyone with even half decent programming skills + is entirely free and open source. This is an advantage in making research reproducible. Although adding Platform IO support won't impact any of this directly it will be a step in the wrong direction.

jeremypoulter, 2018-02-12:
I have and am working on a few commercial projects using Arduino.

Arduino has a huge amount of support for a wide range of chips and
peripherals that the native SDK does not. This makes it an excellent choice
for rapid development and also when combined with Platform IO it makes for
an great choice in the commercial environment.

The reason I say Platform IO (other than being on topic) is that it is just
so easy to get people set up, install the IDE open the project and build,
no messing around finding the right version of a libraries and all the
tools and features you need in a professional environment.

Anyway just my 2p worth.

dlabun, 2018-02-12:
@curtpw I don't want to read your comment incorrectly, are you saying you're against supporting PlatformIO?

sandeepmistry, 2018-02-13:
@ivankravets thank you for starting a discussion to propose changes to the read me. At this time we are not open to the changes discussed, due to the fact they would:
* Imply maintainers actively use this project with PlatformIO
* Double the test matrix for maintenance and releases

In the future, please keep discussions focused, relevant, and constructive. This is not a place to talk negatively and/or spread negativity about other people, communities, and organizations.

closed 2018-02-13 by sandeepmistry

ivankravets, 2018-02-13:
Thanks! We will continue support this project on our side. Users will decide use it or not.