sandeepmistry/arduino-nRF5 #236: sending nRF52 to sleep/low power

mristau, 2018-02-09:
We have a board with nRF52 on it, some laser distance sensors and an accelerometer. This board should work on a LiPo battery. The accelerometer has a pin for interrupts and we would like to send the nRF52 to sleep and the laser sensors power down when no movement has happened for x minutes.

I have read the documentation of the nRF52 there are several ultra low power modes which we would like to use. My problem is just, how would i send it to the chip using this library.

I have just taken this project from my colleague, so for now i have to stick to Arduino and this Library, because it should work soon.

dlabun, 2018-02-09:
At this moment there's no low power mode support in this library. There's already an open request to port over the Arduino Primo LPM, but there hasn't been any recent activity.

sandeepmistry, 2018-02-10:
Closing this as a duplicate of https://github.com/sandeepmistry/arduino-nRF5/issues/168.

closed 2018-02-10 by sandeepmistry

drmpf, 2019-03-06:
I have a mod of this package that supports low power modes see
https://www.forward.com.au/pfod/BLE/LowPower/index.html