sandeepmistry/arduino-nRF5 #238: Setup backer platform?

sandeepmistry, 2018-02-10:
@dlabun any thoughts on setting up something like [OpenCollective](https://opencollective.com) or equivalent for this project?

@jacobrosenthal and myself have [one setup](https://opencollective.com/noble) for [noble](https://github.com/sandeepmistry/noble). No strong opinion on using it, but we haven't had any trouble with it.

Some ideas on what the funds collected could be used for are:
 * any (hardware) expenses occurred during maintenance
 * (if things go well) covering expenses for the active maintainers of the project to meet in person at a tech conference

dlabun, 2018-02-10:
I think that's a great idea... 99% of the time I am answer "I don't own that board" so having a source of funding to actually have hardware would help the project greatly.

sandeepmistry, 2018-02-12:
OpenCollective is setup now.

Read me edits in https://github.com/sandeepmistry/arduino-nRF5/commit/5a142fd07acdcaa36bb03bf09e4a6fd42b67e31d. Let me know if we should move it down and if you have improvements for logo ideas :)

closed 2018-02-12 by sandeepmistry