sandeepmistry/arduino-nRF5 #239: Is the WT51822-S4AT board supported?

probonopd, 2018-02-11:
Is the WT51822-S4AT supported?

http://www.wireless-tag.com/wireless_module/BLE/WT51822-S4AT.html

It seems like P0.03, P0.04, P0.09, P0.10, P0.13 are available. With these, is it possible to use
* i2c
* SPI
devices from within the Arduino environment on this board?

dlabun, 2018-02-11:
Currently it's not supported but you can use the generic nRF51 device and modify it to work with that board.

probonopd, 2018-02-11:
So you think hardware-wise there are no issues?

dlabun, 2018-02-11:
Hard to tell since it's pre-programmed to act as a UART to Bluetooth bridge. I would check with the manufacturer first to see if the board is designed to be reprogrammed.

probonopd, 2018-02-11:
According to the data sheet, it has pins for

|   Pin	|  Signal 	|   Comment	|   
|---	|---	|---	|
|pin10 | SWDIO  | Debug data input|
|pin11 | SWDCLK  | Debug clock signal input|

Is there anything else that needs to be in the spec in order for it to be programmable with this core?

sandeepmistry, 2018-02-11:
That should be it and an external programmer that supports programming via SWDIO/SWCLK.

closed 2018-02-11 by sandeepmistry

tamasharasztosi, 2021-04-11:
Hello! 

Did you managed to make it work? 
Could you please guide me, I would like to do the same thing with that module, I will hook it up with an ST-Link V2, is it going to work?

Thank you!