sandeepmistry/arduino-nRF5 #240: Support DFU over-the-air updates?

probonopd, 2018-02-11:
I know that it was [stated earlier](https://github.com/sandeepmistry/arduino-nRF5/issues/32#issuecomment-241288101) that there is no support for DFU in this project. But possibly things may have changed, now that there seem to be open source tools available?

* https://github.com/redbear/nRF5x seems to have the required bootloader
* https://github.com/redbear/nRF5x/blob/master/nRF52832/docs/DFU_User_Guide.md says that you can export an Arduino sketch from the Arduino IDE and load it into nrfutil.exe to generate an update file that can be used with the nRF Toolbox App
* https://github.com/NordicSemiconductor/pc-nrfutil appears to be an open source version of this tool

sandeepmistry, 2018-02-11:
@probonopd what are you proposing? Maybe opening a pull request would be the best way to continue the discussion.

probonopd, 2018-02-11:
I am not actually sure what is needed to make DFU work with this project. Ordered a module now, I'll experiment with it. Just thinking that DFU would be so much nicer than fiddling with SWDIO and SWDCLK during development all the time.

sandeepmistry, 2018-02-12:
Ok, let us know how your experience with DFU goes. User's will still need to use SWDIO/SWDCLCK to get going.

closed 2018-02-12 by sandeepmistry