sandeepmistry/arduino-nRF5 #241: Programming requirements after flashing SoftDevice

YuriPL2, 2018-02-12:
Hi,

I am on design stage of device using NRF51822 and I am planning to use Your NRF5 Arduino Core. I  will embedd programmer on board and have aquestion about programming procedure, do both SoftDevice and Sketch are uploaded via JTAG, or after flashing SoftDevice the sketch can be uploaded via serial? 

Best regards

dlabun, 2018-02-12:
Everything is programmed via SWD... What kind of programmer are you embedding? Just curious because JLink OB is the only onboard debugger I've seen outside of Atmel EDBG.

YuriPL2, 2018-02-12:
Thanks for response,
I was asking because I am thinking about what programmer should I use. This is just prototype, so for now I am thinking on using STM32 to implement Blackmagick Probe. Can I use Blackmagick Probe directly from Arduino IDE? I don't see it under "programmers" menu.

dlabun, 2018-02-12:
We added support for the Black Magic Probe a few months ago... You'll see it as an option in the next release. I should have the next release posted here pretty soon.

dlabun, 2018-02-17:
BMP support is now available via the 0.5.0 release.

closed 2018-02-17 by dlabun

probonopd, 2018-02-17:
> Can I use Blackmagick Probe directly from Arduino IDE?

Yes. I can confirm that it works on Linux with my [patch](https://github.com/sandeepmistry/arduino-nRF5/pull/247).