sandeepmistry/arduino-nRF5 #243: Interrupt to wakeup/reset from power off

mristau, 2018-02-13:
I've managed to put my nRF52 Board to power off mode, but an interrupt i added with attachInterrupt works while it is running, not when it's in power off mode.
I can see a current spike from 0.09mA to 0.220mA when i press the button.
I included WInterrupts.h, nrf_soc.h and nrf_sdm.h already from the package

The button is setup with pinMode(BTN_PIN, INPUT_PULLUP)
The interrupt is setup with attachInterrupt(digitalPinToInterrupt(BTN_PIN), interruptHandler, FALLING);

I looked in the ArduinoLowPower Library, they use following function to attach an interrupt for wakeup. Seems similar to what i found in the nordic sdk
nrf_gpio_cfg_sense_input(g_APinDescription[pin].ulPin, NRF_GPIO_PIN_PULLUP, NRF_GPIO_PIN_SENSE_LOW);

I couldn't find this function anywhere in this package, can somebody point me to where i can find it, or what is it called here

mristau, 2018-02-16:
After research i finally found a way to enable the Sense with following command

`NRF_GPIO->PIN_CNF[LIS2DE12_INT1_PIN] |= ((uint32_t)GPIO_PIN_CNF_SENSE_High << GPIO_PIN_CNF_SENSE_Pos);`

Standard is SENSE_Disabled, other possible is SENSE_Low, maybe i'd have to reset the bitfield first to make sure only the desired method is active

Is there any better way to do this, maybe a library method?

dlabun, 2018-02-16:
You would need to write your own library for the time being.

probonopd, 2018-02-17:
I would also be interested in this, but for the nRF51.

mristau, 2018-02-18:
It should be the same for nRF51

I found the code in the pinMode(..) implementation in wiring_digital.c
Informations on the bits to set and the defines are in nrf5x_bitfields.h

Currently i'm sending my board to power off with a button press, but it will be later with lack of movement for some time. Wakeup is by an interrupt from a lis2de12 accelerometer

probonopd, 2018-02-18:
Thank you for your pointers @mristau, especially at https://github.com/sandeepmistry/arduino-nRF5/issues/168#issuecomment-366488321. Do you think you could make a small example sketch from it? (I am lacking the proper current measurement circuitry at the moment so I'd have to work "blindly".)

mristau, 2018-02-19:
```
  NRF_GPIO->PIN_CNF[INTERRUPT_PIN] &= ~((uint32_t)GPIO_PIN_CNF_SENSE_Msk);
  NRF_GPIO->PIN_CNF[INTERRUPT_PIN] |= ((uint32_t)GPIO_PIN_CNF_SENSE_High << GPIO_PIN_CNF_SENSE_Pos);
```
this should make everything, first resetting the bits, then setting to the right value High/Low is possible

sandeepmistry, 2018-02-22:
I'd suggest making a new Arduino library for this suggestion.

closed 2018-02-22 by sandeepmistry