sandeepmistry/arduino-nRF5 #244: Uploading code to nRF52832

Xmodgeek, 2018-02-13:
I've been using the nRF51422 for awhile now, but I'm now moving to the nRF52810 and having issues uploading anything. Including the nRF52832 that's in the 52DK. I'm unsure if the library here supports the nRF52810 but for now I'm using the nRF52832 just to get started with the 52 series. 

So far what I've been doing is using the nRFgo Studio program to upload the SoftDevice then the application.
I get a error saying "This hex file has data in SoftDevice region. Try programming using "Program SoftDevice", or erase all before programming."

I've tried using the Generic nRF52 and the Nordic Semiconductor nRF52 DK board.
the s132 Soft Device is selected, it won't compile otherwise. 
I've tried to upload my program and the BLEPeripheral Serial example program to the same effect.

I have used the 52DK to upload to my custom hardware using the nRF51422 So I'm hoping its a software thing...

Thanks

dlabun, 2018-02-13:
There shouldn't be any reason to use nRFgo. Please try it flashing everything from the Arduino IDE.

Xmodgeek, 2018-02-14:
Thanks a lot that fixed it. I liked to use nRFgo so I could still try programing using the Nordic code with out having to mess with drivers all the time. Is there a reason exporting the HEX file works with the nRF51 but not the nRF52?

dlabun, 2018-02-16:
I think the issue is caused by an extra line return in the HEX file when it's export by the Arduino IDE. If you were to open the exported HEX file and check it you will probably see it at the very end of the file. Delete it and nRFgo should work okay.

That's assuming it's the extra line return issue...

Xmodgeek, 2018-02-16:
I tried that and it didn't work. The HEX for the nRF51 has the extra line return as well, but that uploads fine.

dlabun, 2018-02-17:
Odd... Not sure what would be causing the issue then. For this project we only really support uploading via the IDE so I am going to close this issue as resolved.

closed 2018-02-17 by dlabun