sandeepmistry/arduino-nRF5 #245: Black Magic Probe (GDB): \\.\/dev/ttyACM0: No such file or directory

probonopd, 2018-02-16:
Have flashed Black Magic Probe firmware to a Blue Pill board as described on http://www.stm32duino.com/viewtopic.php?f=37&t=2801#p36730:

```
stm32flash -w blackmagic.bin -S 0x08000000 -v /dev/ttyUSB0
```

The system does see the device:

```
$ cat /etc/*release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=16.04

$ lsusb
Bus 008 Device 012: ID 1d50:6018 OpenMoko, Inc. 

$ dmesg
[16845.933046] usb 8-1: Product: Black Magic Probe (STLINK), (Firmware v1.6.1-101-g5686550-dirty)
[16845.933048] usb 8-1: Manufacturer: Black Sphere Technologies
[16845.933051] usb 8-1: SerialNumber: ...
[16845.939092] cdc_acm 8-1:1.0: ttyACM0: USB ACM device
[16845.942083] cdc_acm 8-1:1.2: ttyACM1: USB ACM device

$ ls -lh /dev/ttyACM*
crw-rw-rw- 1 root dialout 166, 0 Feb 16 13:23 /dev/ttyACM0
crw-rw-rw- 1 root dialout 166, 1 Feb 16 13:23 /dev/ttyACM1
```

However, if I try to upload a sketch using the __Black Magic Probe (GDB)__ programmer from the Arduino IDE, I get:

```
Sketch uses 1296 bytes (0%) of program storage space. Maximum is 262144 bytes.
\\.\/dev/ttyACM0: No such file or directory.
"monitor" command not supported by this target.
Don't know how to attach.  Try "help target".
You can't do that when your target is `exec'
Don't know how to run.  Try "help target".
Reading symbols from sketch_feb16a.ino.elf...done.
Temporary breakpoint 1 at 0x322: file /home/me/.arduino15/packages/sandeepmistry/hardware/nRF5/0.5.0/cores/nRF5/main.cpp, line 29.

Program complete!
```

I don't think it has actually done anything since it could not find the device.

Note the extra `\\.\` in `\\.\/dev/ttyACM0: No such file or directory`. @sandeepmistry also has  the extra `\\.\` in https://github.com/sandeepmistry/arduino-nRF5/pull/196#issuecomment-346187018.

probonopd, 2018-02-16:
If I delete all `\\.\` from `platform.txt` and `programmers.txt`, then I do __not__ get this error.

Then I get:

```
Sketch uses 1296 bytes (0%) of program storage space. Maximum is 262144 bytes.
Target voltage: unknown
SW-DP scan failed!
Remote debugging using /dev/ttyACM0
Attaching to Remote target
Attaching to Remote target failed
Reading symbols from sketch_feb16a.ino.elf...done.
Loading section .text, size 0x510 lma 0x0
Temporary breakpoint 1 at 0x322: file /home/me/.arduino15/packages/sandeepmistry/hardware/nRF5/0.5.0/cores/nRF5/main.cpp, line 29.
Starting program: /tmp/arduino_build_965627/sketch_feb16a.ino.elf 

Program complete!
Load failed
Running the default executable on the remote target failed; try "set remote exec-file"?
```

Which I am not sure whether this has to do with my hardware setup.

dlabun, 2018-02-16:
By any chance do you have an actual black magic probe to test with? I don't own either device to test with unfortunately. I am just wondering if this anything to do with the fact that the BMP firmware has been recompiled for a different board.

probonopd, 2018-02-16:
Working for me now. So at least on Linux, the `\\.\` definitely needs to be removed.

dlabun, 2018-02-16:
@sandeepmistry can you take a look at this? I don't have a clue when it comes to Linux

probonopd, 2018-02-16:
These Blue Pill boards can be had for around $2 and might become quite popular as programmers with the BMP firmware. Performing very well for me now, with the change described above.

mristau, 2018-02-19:
From the BlackMagic Readme

> Note: On Mac OS X use the /dev/cu.usbmodem<somenumber>. Don't use the tty version of this device as this one will freeze gdb when it tries to open it.

> On Windows, use the the COM port instead of /dev/ttyACM0. For ports >= COM10, add the prefix \\.\, for example:

```
target extended-remote COM3
target extended-remote \\.\COM10
```

probonopd, 2018-02-19:
While `\\.\` should be used on Windows with COM-Ports higher than 10, the `\\.\` entirely prevents this from working from anything but Windows. No other core uses the `\\.\` Windows workaround.

Please merge my pull request. The documentation states that `\\.\` needs to be used on Windows with COM-Ports higher than 10. But it should not be the default.

probonopd, 2018-02-21:
How do other Arduino cores handle this? How common is it to have a 2-digit COM port in Windows? I haven’t encountered this topic with other cores so far that also upload via the serial port (e.g., Arduino Uno).

dlabun, 2018-02-21:
Honestly, I don't use the BMP or Linux so my opinion how to fix the problem isn't valid... I do know 2 digit COM ports are rather common on Windows, my Arduino Zero is up in the COM60 range.

mristau, 2018-02-21:
windows keeps track of many COM-devices, we develop a board and it's fairly common that first boards start at COM3-4 but i have some that are also high up like COM30-40

probonopd, 2018-02-21:
That could be the solution then

dlabun, 2018-02-21:
If somebody can write a PR and test it, I would be available to issue an urgent release this weekend.

probonopd, 2018-02-21:
Added lines for Windows to my PR at https://github.com/probonopd/arduino-nRF5/commit/b8a3b1a70a9d0dd0fd5eb377a95dc17a5a5d2903 - would be great if someone with Windows could test this.

dlabun, 2018-02-21:
Before I merge it I would like confirmation it works on all 3 platforms.

micooke, 2018-02-21:
I can test this tomorrow

micooke, 2018-02-23:
Confirmation for Windows :ok_hand:

dlabun, 2018-02-23:
If no issues are found between now and Sunday I’ll issue a new release around noon Sunday.

dlabun, 2018-02-24:
0.5.1 is available as a pre-release on Github, I'll make it available in the Arduino IDE tomorrow.

dlabun, 2018-02-27:
0.5.1 is now live in the Board Manager, Travis is passing as well.

closed 2018-02-27 by dlabun

probonopd, 2018-03-01:
Thank you @dlabun.

suibin0127, 2020-02-29:
Hi，i am very interesting to learn from this practical experience to convert a STM32 blue board to BMP debugger. I bought one but looks i have trouble to program it using USART, after setting right Boot Mode and RS232 Converter (PL2303), i saw "Init Device Fail.".. tried STM32 official Flash Loader and Ardurino STM32 Flash Loader, all the same. So, I have to choose J-Link to download the BMP image. Following is he Downloading process, it successfully.
But after that, I can't see a USB attachment looks firmware doesnt work properly.
The VM centos didnt detect a USB device, but Jlink can detect.
I think there must somehting wrong with the programming process.
I tried following BMP bin image, same. Could anybody help me to address this problem ?
**blackmagic_dfu-native.bin
blackmagic_dfu-stlink.bin**

> [root@bogon src]# lsusb
> Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
> Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
> Bus 002 Device 002: ID 0e0f:0003 VMware, Inc. Virtual Mouse
> Bus 002 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
> Bus 002 Device 004: ID 0e0f:0008 VMware, Inc.

 

> - Start of flash programming
>  - Programming range 0x08000000 - 0x08001FFF (  8 Sectors, 8 KB)
>  - End of flash programming
>  - Flash programming performed for 1 range (8192 bytes)
>  - 0x8000000 - 0x8001FFF (  8 Sectors, 8 KB)
>  - Start of verifying flash
>  - End of verifying flash
>  - Start of restoring
>  - End of restoring
>  - Target programmed and verified successfully (CRC = 0x14D60F0F) - Completed after 0.434 sec

suibin0127, 2020-02-29:
> Hi，i am very interesting to learn from this practical experience to convert a STM32 blue board to BMP debugger. I bought one but looks i have trouble to program it using USART, after setting right [...]

the 2 bin after flashing show red led, no blink
i tried "blackmagic_dfu-swlink.bin", it have green blink, but usb still not recognize.
i suppose swlink one is the working image for this board.

Then i switch to Ardunio IDE using BMP to upload a program to test it via COM4
but failed. the board info get as below, 
BN: Unknnown Board
VID: 067B
PID: 2303
SN: Upload any sketch to obtain it

micooke, 2020-02-29:
It sounds like a BMP version issue, rather than a nrf5x issue. Can you confirm that your BMP works with other devices? Have you flashed the bootloader prior to your fitst flash? Etc etc

suibin0127, 2020-02-29:
> It sounds like a BMP version issue, rather than a nrf5x issue. Can you confirm that your BMP works with other devices? Have you flashed the bootloader prior to your first flash? Etc etc

My bought board bootloader looks doesnt work fine for USART init, so I use Jlink erease all flash and then do the BMP bin programming direclty. Is that mean the BMP bin need program base one certain bootloader? I just use SWD interace to program Flash Memory start @0x80000000
and I dont have other device in hand since I  am quite new to STM32 only play with the blue board.

so, is the step to try as below correct? The flash starting address are all 0x80000000 ?
thanks !

1. download form thsi bootloader and flash first
  https://github.com/bootsector/stm32-hid-bootloader
2. then flash BMP

micooke, 2020-02-29:
Ok i misread your previous post.
There are lots and lots of guides online on how to flash a "blue pill" as a bmp. I just used a usb ttl, ez. It sounds like either your bit file is too big/wrong, or that you arent setting the boot mode properly, or that you simply need to hard reset your stm32. Having said this, I'm sorry but this is not the right place for help. Please follow a guide, ask the author if you have issues, or contact blacksphere.
I even wrote a gist once apon a time, but it is probably very outdated https://gist.github.com/micooke/e9a4cf831316ace10ee6506ebbab95ed

suibin0127, 2020-02-29:
> Ok i misread your previous post.
> There are lots and lots of guides online on how to flash a "blue pill" as a bmp. I just used a usb ttl, ez. It sounds like either your bit file is too big/wrong, or [...]

Appreciate for help! I  will study on it with the info you provided.

suibin0127, 2020-03-01:
[image]
For further study, I know need to flash 2 BMP bin image in Flash Memory (0x08000000~0x0801FFFF)
"blackmagic_dfu-swlink.bin"  starting @ 0x08000000  (8KB)
"blackmagic-swlink.bin" starting @0x080020000 (80KB)

After that, i reset MCU and suppose to the board simulate a USB emulator for JTAG & SWD..
But system have no response.  Then I use Jlink to dump the System BOOT memory
(0x00000000~0x7FFFFFFF), the  content is blank as below.

So, I think why my attempts to use USART to flash BMP doesnt work, since even I set BOOT mode to USART, there has no BOOTP inside this memory after device init.
Then, quesiton is how to recover this area, maybe I need have a dump image which i can reprogram back the BOOTP area or i use keil USART App to flash this area?

[image]

commit 33ae4d5c (2018-02-24): Remove '\\.\', closes #245