sandeepmistry/arduino-nRF5 #246: Remove the need for 32-bit libraries on Linux

probonopd, 2018-02-17:
The README says:

> For 64-bit Linux users,  ```libc6:i386```, ```libstdc++6:i386```, ```libncurses5:i386``` and ```libudev1:i386``` need to be installed :
   * ```sudo dpkg --add-architecture i386```
   * ```sudo apt-get update```
   * ```sudo apt-get install libc6:i386 libstdc++6:i386 libncurses5:i386 libudev1:i386```

Please use native 64-bit applications for everything, then the 32-bit compatibility libraries will not be needed.

probonopd, 2018-02-17:
`arm-none-eabi-g++` seems to be available for 64-bit Linux from https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads - can that one be used?

dlabun, 2018-02-17:
Until we upgrade to a new version of the toolchain the 32-bit version is required.

closed 2018-02-17 by dlabun

probonopd, 2018-02-17:
What prevents the project from upgrading to the newest toolchain now?

dlabun, 2018-02-17:
Nothing other than the time and effort to upgrade and test the new toolchain on all 3 platforms

probonopd, 2018-02-17:
Would it be possible to leave this ticket open as a reminder, until it is solved? Thanks.

sandeepmistry, 2018-02-26:
Once #106 is done, this will be solved.