sandeepmistry/arduino-nRF5 #253: Please enable the Wiki

probonopd, 2018-02-19:
Please enable https://github.com/sandeepmistry/arduino-nRF5/wiki and let users edit, so that we can collect documentation (e.g., about boards and pinouts) there. Thanks.

sandeepmistry, 2018-02-22:
@dlabun any thoughts on this?

The wiki is already enabled from what I see.

dlabun, 2018-02-24:
Having a wiki isn't a bad idea, I am just concerned about having to police it for vandalism. Can it be modified through the PR process?

probonopd, 2018-02-24:
I don't think so but from my experience people on GitHub are generally civilized. In case it turns out this isn't the case, you can limit edit rights to collaborators; but I think this would take away from the power of a wiki.

sandeepmistry, 2018-02-26:
I'm up for trying the wiki. We can always disable it if things don't work out.

I've started a welcome page: https://github.com/sandeepmistry/arduino-nRF5/wiki

closed 2018-02-26 by sandeepmistry