sandeepmistry/arduino-nRF5 #255: Support for nRF52810

Xmodgeek, 2018-03-01:
Is there currently or planed any support for the nRF52810? I've started working with them and had no luck getting it to work out of the box (kinda figured as much) 

I got 'upload' to work but failed on the softdevice.
Seams S112 is what's intended for the nRF52810.

I saw this looking in the nRF52810 documentation 

 The S132 5.0.0-3.alpha is intended to be run on the nRF52832 IC. It may be used on the nRF52810 IC for development purposes. It should not be used for production on the nRF52810 IC.

Is changing the softdevice that's programed as simple as changing the URL and filename in the softdevices.txt?

dlabun, 2018-03-01:
The nRF52810 isn't currently supported, it requires a much newer SDK and soft device than what this project is based on. You may be able to flash the soft device by modifying the project but there's no guarantee the code you create will work right.

sandeepmistry, 2018-03-11:
Pull requests welcome for this, I think mainly the linker files need to be changed.

sandeepmistry, 2020-12-02:
PR #493 includes an update MDK with a linker file for nRF52810 now.

gabewillen, 2021-01-09:
I’ll be attempting to add support for S112 or S113 today. I have an NRF52811 that will not
work with the three soft devices available to choose from.