sandeepmistry/arduino-nRF5 #256: Cant flash softdevice

yldzmuhammed, 2018-03-04:
Hi all,

I have followed the instructions on the main page.But i could not flash softdevice yet.

Here is what message i get when i try to flash it;

`Arduino:1.8.5 (Windows Store 1.8.10.0) (Windows 10), Kart:"Generic nRF52, S132, Crystal Oscillator"

Open On-Chip Debugger 0.10.0-dev-00254-g696fc0a (2016-04-10-10:13)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 0
0
adapter speed: 10000 kHz
cortex_m reset_config sysresetreq
in procedure 'init' 
in procedure 'ocd_bouncer'

Error while flashing SoftDevice.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`
My board is Generic nRF54 softdevice is S132 and i am using my nRF52832QFAA board.

What should i do? How can i fix this?

dlabun, 2018-03-04:
What board are you using? If it's not an mbed board you won't be able to install the driver.

dlabun, 2018-03-07:
Closing due to no response

closed 2018-03-07 by dlabun