sandeepmistry/arduino-nRF5 #257: hardware Timer

RezApp, 2018-03-06:
i'm trying to use hardware timer2 with this example
[https://github.com/NordicPlayground/nrf51-TIMER-examples/blob/master/timer_example_timer_mode/main.c](url)
but TIMER2_IRQHandler never runs
what should i do?

dlabun, 2018-03-07:
Your best bet is to ask on the Nordic forums for help with their code example.

RezApp, 2018-03-07:
i know but i need to use timer in arduino ide
is there any way to use timer2?

sandeepmistry, 2018-03-11:
@RezApp please share the sketch you are using.

RezApp, 2018-03-11:
@sandeepmistry here you are
i used NRF51822 QFAAH1 chip with internal RC oscillator
`#define LED_PIN 19

void TIMER2_IRQHandler(void)
{
  if ((NRF_TIMER2->EVENTS_COMPARE[0] != 0) && ((NRF_TIMER2->INTENSET & TIMER_INTENSET_COMPARE0_Msk) != 0))
  {
    NRF_TIMER2->EVENTS_COMPARE[0] = 0;           //Clear compare register 0 event
    digitalWrite(LED_PIN, HIGH);                 //Set LED
  }

  if ((NRF_TIMER2->EVENTS_COMPARE[1] != 0) && ((NRF_TIMER2->INTENSET & TIMER_INTENSET_COMPARE1_Msk) != 0))
  {
    NRF_TIMER2->EVENTS_COMPARE[1] = 0;           //Clear compare register 1 event
    digitalWrite(LED_PIN, LOW);                  //Clear LED
  }
}

void start_timer(void)
{
  NRF_TIMER2->MODE = TIMER_MODE_MODE_Timer;  // Set the timer in Counter Mode
  NRF_TIMER2->TASKS_CLEAR = 1;               // clear the task first to be usable for later
  NRF_TIMER2->PRESCALER = 6;                             //Set prescaler. Higher number gives slower timer. Prescaler = 0 gives 16MHz timer
  NRF_TIMER2->BITMODE = TIMER_BITMODE_BITMODE_16Bit;     //Set counter to 16 bit resolution
  NRF_TIMER2->CC[0] = 25000;                             //Set value for TIMER2 compare register 0
  NRF_TIMER2->CC[1] = 5;                                 //Set value for TIMER2 compare register 1

  // Enable interrupt on Timer 2, both for CC[0] and CC[1] compare match events
  NRF_TIMER2->INTENSET = (TIMER_INTENSET_COMPARE0_Enabled << TIMER_INTENSET_COMPARE0_Pos) | (TIMER_INTENSET_COMPARE1_Enabled << TIMER_INTENSET_COMPARE1_Pos);
  NVIC_EnableIRQ(TIMER2_IRQn);

  NRF_TIMER2->TASKS_START = 1;               // Start TIMER2
}

void setup() {
  pinMode(LED_PIN, OUTPUT);
  start_timer();
}

void loop() {
}`

RezApp, 2018-03-11:
i've just added 
```c++
#ifdef __cplusplus
extern "C" {
#endif
void TIMER2_IRQHandler(void);
#ifdef __cplusplus
}
#endif
```
at the top of my code and it is working nice

sandeepmistry, 2018-03-12:
Cool

closed 2018-03-12 by sandeepmistry