sandeepmistry/arduino-nRF5 #258: Update to SoftDevice 5.1.0

jemertz, 2018-03-10:
The softdevice 132 has been updated to 5.1.0 and marked as Production
http://www.nordicsemi.com/eng/nordic/Products/nRF52832/S132-SD-v5/58987

I see that S132 v5.1.0 is API compatible with the S132 v5.0.0, but everything I have looked at indicates that this package is using v2.0.1. Will updating this break API compatibility? If not, I'd like to see it happen.

dlabun, 2018-03-10:
You can certainly try updating to a newer SD, but we wouldn't be able to support any issues that occur as that SD requires a much new version of the Nordic SDK.

closed 2018-03-10 by dlabun

sandeepmistry, 2018-03-11:
@jemertz best way to get this in is to submit a pull request. You need to update:

* [cores/nRF5/SDK/components/softdevice/s132](https://github.com/sandeepmistry/arduino-nRF5/tree/master/cores/nRF5/SDK/components/softdevice/s132) with new files
* [softdevices.txt](https://github.com/sandeepmistry/arduino-nRF5/blob/master/softdevices.txt) with new link and filename

It will have to be a "breaking" version change for the core release, then arduino-BLEPeripheral will also need to be updated to use the new soft device API's.