sandeepmistry/arduino-nRF5 #262: Application start address did not match with the base address of code region 1

sashisekhar, 2018-03-21:
hello sandeep,
i am a newbie in nrf51822. i am using your libraries for programming. but when i try to load one of your examples(test.ino) on my ble400 board then its showing some fatal error. i hv been trying this with s110 & s130 both. but same result.  the error i am getting from nrfgostudio is "Running application C:/Users/SASHIS~1/AppData/Local/Temp/nrf51Studio/nrfjprog.exe --snr 123456 --program C:/Users/sashisekhar/Documents/Arduino/code/test/test.ino.Waveshare_BLE400.hex --reset
program base address is 0X18000  code region 1 base address = 0X14000
ERROR: Application start address did not match with the base address of code region 1 ".
please guide me how to resolve the issue. 
thanks in advance

dlabun, 2018-03-21:
The nRFStudio application will not work properly with the Hex file generated by the Arduino IDE. As long as you're using one of the support programmers you can flash your board directly from the Arduino IDE.

sashisekhar, 2018-03-21:
thanks @dlabun for the quick reply. i have the jlink-ob programmer which is recommended by the waveshare itself. is it a recommended programmer for arduino or should i go with any other programmer.

dlabun, 2018-03-21:
The JLink will work fine, just follow the directions in the readme file to set it up.

sashisekhar, 2018-03-22:
hiii @dlabun,
i have tried several times but the same error is comming like
"Open On-Chip Debugger 0.10.0-dev-00254-g696fc0a (2016-04-10-10:13)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
0
cortex_m reset_config sysresetreq
adapter speed: 1000 kHz
An error occurred while uploading the sketch
jaylink: Failed to open device: LIBUSB_ERROR_NOT_SUPPORTED.
Info : No device selected, using first device.
Error: No J-Link device found.
Error: No Valid JTAG Interface Configured."
how can i resolve this issue? 
if u can provide me some tutorial or file through which i can set it up properly then it willl be a great help for me

dlabun, 2018-03-22:
Did you follow the directions on the very start of this project to install the JLink driver?

sashisekhar, 2018-03-23:
hello,
i found the issue. as per the waveshare wikipedia i have downloaded older versions of sdk & go studio. thats why the address issue arrises. once i updated it to the latest versions all work perfectly fine. thank u

closed 2018-03-23 by dlabun