sandeepmistry/arduino-nRF5 #264: documentation request: how to physically connect

flatsiedatsie, 2018-04-06:
I was wondering if it would be possible to expand the readme a little bit, and cover the proces all the way to a blink sketch. This could answer beginners questions like:
- Can a normal usb-to-serial module be used?
- To what pins should the usb-to-serial device be connected?
- Is there anything (other commands) that the blink sketch should contain?
- Etc

Something that also isn't completely clear to me: if I don't want to use bluetooth, does that make the process simpler?

cimba007, 2018-04-10:
@dlabun I am struggeling here a bit too. I connected stlink-v2 and successfully flashed the softdevice.

```
C:\Users\Marc\AppData\Local\Arduino15\packages\sandeepmistry\tools\openocd\0.10.0-dev.nrf5/bin/openocd.exe -d2 -f interface/stlink-v2.cfg -c transport select hla_swd; set WORKAREASIZE 0x4000; -f target/nrf51.cfg -c init; halt; nrf51 mass_erase; program {{C:\Users\Marc\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.5.1/cores/nRF5/SDK/components/softdevice/s110/hex/s110_nrf51_8.0.0_softdevice.hex}} verify reset; shutdown; 
Open On-Chip Debugger 0.10.0-dev-00254-g696fc0a (2016-04-10-10:13)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
0x4000
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 1000 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Info : STLINK v2 JTAG v21 API v2 SWIM v4 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.155992
Info : nrf51.cpu: hardware has 4 breakpoints, 2 watchpoints
nrf51.cpu: target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x61000000 pc: 0x000164ca msp: 0x20001500
Info : nRF51822-QFAB(build code: C0) 128kB Flash
nrf51.cpu: target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0xc1000000 pc: 0xfffffffe msp: 0xfffffffc
** Programming Started **
auto erase enabled
Info : Padding image section 0 with 2112 bytes
Warn : using fast async flash loader. This is currently supported
Warn : only with ST-Link and CMSIS-DAP. If you have issues, add
Warn : "set WORKAREASIZE 0" before sourcing nrf51.cfg to disable it
nrf51.cpu: target state: halted
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000001e msp: 0xfffffffc
wrote 93184 bytes from file C:\Users\Marc\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.5.1/cores/nRF5/SDK/components/softdevice/s110/hex/s110_nrf51_8.0.0_softdevice.hex in 3.935076s (23.125 KiB/s)
** Programming Finished **
** Verify Started **
nrf51.cpu: target state: halted
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000002e msp: 0xfffffffc
nrf51.cpu: target state: halted
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000002e msp: 0xfffffffc
verified 90328 bytes in 0.613771s (143.720 KiB/s)
** Verified OK **
** Resetting Target **
shutdown command invoked
```

For now this seems fine and I see no error .. but what is next? How do I use the "bootloader" .. Do I need a USB-Serial adapter and if so what pins do I have to connect? I am using a rf51822 from ebay which doesn't match any of the listed boards.

dlabun, 2018-04-10:
There is no bootloader, everything is programmed directly through the programmer itself. From the looks of the second error your programmer has simply stopped communicating, did you still have it connected?

cimba007, 2018-04-10:
The first piece of code was uploaded using the "nRF5 Flash SoftDevice" option. Is this the right option to flash everything?

cimba007, 2018-04-10:
For now my assumtion is that the generic rf51 is missing

Generic_nRF51822.upload.protocol=
Generic_nRF51822.upload.setup_command=

cimba007, 2018-04-10:
Blinki Blinki ..

```
/*
  Blink

  Turns an LED on for one second, then off for one second, repeatedly.

  Most Arduinos have an on-board LED you can control. On the UNO, MEGA and ZERO
  it is attached to digital pin 13, on MKR1000 on pin 6. LED_BUILTIN is set to
  the correct LED pin independent of which board is used.
  If you want to know what pin the on-board LED is connected to on your Arduino
  model, check the Technical Specs of your board at:
  https://www.arduino.cc/en/Main/Products

  modified 8 May 2014
  by Scott Fitzgerald
  modified 2 Sep 2016
  by Arturo Guadalupi
  modified 8 Sep 2016
  by Colby Newman

  This example code is in the public domain.

  http://www.arduino.cc/en/Tutorial/Blink
*/

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(29, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
    digitalWrite(29, HIGH);   // turn the LED on (HIGH is the voltage level)
    delay(1000);                       // wait for a second
    digitalWrite(29, LOW);    // turn the LED off by making the voltage LOW
    delay(1000);                       // wait for a second
}
```

on this board:
 https://de.aliexpress.com/item/2V-3-3V-NRF51822-Bluetooth-4-0-Wireless-Module-For-iBeacon-Base-Station-Intelligent-Control-System/32824004498.html

with
```
Generic_nRF51822.upload.protocol=stlink-v2
Generic_nRF51822.upload.setup_command=transport select hla_swd;
```

dlabun, 2018-04-10:
The generic device is just a template, it's up to the you to really customize it to whatever non-support board you are using. I am little puzzled as why you had to add those lines, if you select a programmer from the list the IDE should have taken care of those settings.

cimba007, 2018-04-11:
Have a close look at this:

C:\Users\Marc\AppData\Local\Arduino15\packages\sandeepmistry\tools\openocd\0.10.0-dev.nrf5/bin/openocd.exe -d2 -f interface/**{upload.protocol}**.cfg -c **{upload.setup_command}** -f target/nrf51.cfg -c program **{{C:\Users\Marc\AppData\Local\Temp\arduino_build_453879/iBeacon.ino.hex}}** verify reset; shutdown; 
Open On-Chip Debugger 0.10.0-dev-00254-g696fc0a (2016-04-10-10:13)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
embedded:startup.tcl:60: Error: Can't find interface/**upload.protocol**.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 60
Der ausgewählte serielle Port at file "embedded:startup.tcl", line 60
 ist nicht vorhanden oder das Board ist nicht angeschlossen

I suspect that the variables for some reason do not get replaced properly .. the curlybrackets get me worried.

cimba007, 2018-04-11:
I am using the current version from the repository: https://sandeepmistry.github.io/arduino-nRF5/package_nRF5_boards_index.json

I will double check when I am home.

dlabun, 2018-04-11:
When you are able, please open a new issue just so we can track your issue properly.... Something isn't right here.

akkaman007, 2018-05-05:
HI i am using ST- Link v2 to program nrf51822(https://www.amazon.in/xcluma-Bluetooth-Module-Nrf51822-Development/dp/B073DD94Q5?tag=googinhydr18418-21). But its not working.The problem which i am facing is that the  softdevice is not flashing.

dlabun, 2018-05-07:
Closing, no response from the OP.

closed 2018-05-07 by dlabun