sandeepmistry/arduino-nRF5 #265: Question/Clarification on using generic rf51

cimba007, 2018-04-11:
As mentioned [here](https://github.com/sandeepmistry/arduino-nRF5/issues/264) I am having problems getting an genric rf51 to work.

I am using 

- Arduino IDE 1.8.5
- BLEPeripheral 0.4.0
- arduino-nRF5 0.5.1

My Programmer is set to ST-Link v2 (chinese version)
Softdevice is set to S110

This is my current boards.txt
```
# nRF51 variants
###################

Generic_nRF51822.name=Generic nRF51

Generic_nRF51822.upload.tool=sandeepmistry:openocd
Generic_nRF51822.upload.target=nrf51
Generic_nRF51822.upload.maximum_size=262144
#Generic_nRF51822.upload.protocol=stlink-v2
#Generic_nRF51822.upload.setup_command=transport select hla_swd;

Generic_nRF51822.bootloader.tool=sandeepmistry:openocd

Generic_nRF51822.build.mcu=cortex-m0
Generic_nRF51822.build.f_cpu=16000000
Generic_nRF51822.build.board=GENERIC
Generic_nRF51822.build.core=nRF5
Generic_nRF51822.build.variant=Generic
Generic_nRF51822.build.variant_system_lib=
Generic_nRF51822.build.extra_flags=-DNRF51
Generic_nRF51822.build.float_flags=
Generic_nRF51822.build.ldscript=nrf51_{build.chip}.ld

Generic_nRF51822.menu.chip.xxaa=16 kB RAM, 256 kB flash (xxaa)
Generic_nRF51822.menu.chip.xxaa.build.chip=xxaa
Generic_nRF51822.menu.chip.xxac=32 kB RAM, 256 kB flash (xxac)
Generic_nRF51822.menu.chip.xxac.build.chip=xxac

Generic_nRF51822.menu.softdevice.none=None
Generic_nRF51822.menu.softdevice.none.softdevice=none
Generic_nRF51822.menu.softdevice.none.softdeviceversion=

Generic_nRF51822.menu.softdevice.s110=S110
Generic_nRF51822.menu.softdevice.s110.softdevice=s110
Generic_nRF51822.menu.softdevice.s110.softdeviceversion=8.0.0
Generic_nRF51822.menu.softdevice.s110.upload.maximum_size=151552
Generic_nRF51822.menu.softdevice.s110.build.extra_flags=-DNRF51 -DS110 -DNRF51_S110
Generic_nRF51822.menu.softdevice.s110.build.ldscript=armgcc_s110_nrf51822_{build.chip}.ld

Generic_nRF51822.menu.softdevice.s130=S130
Generic_nRF51822.menu.softdevice.s130.softdevice=s130
Generic_nRF51822.menu.softdevice.s130.softdeviceversion=2.0.1
Generic_nRF51822.menu.softdevice.s130.upload.maximum_size=151552
Generic_nRF51822.menu.softdevice.s130.build.extra_flags=-DNRF51 -DS130 -DNRF51_S130
Generic_nRF51822.menu.softdevice.s130.build.ldscript=armgcc_s130_nrf51822_{build.chip}.ld

Generic_nRF51822.menu.lfclk.lfxo=Crystal Oscillator
Generic_nRF51822.menu.lfclk.lfxo.build.lfclk_flags=-DUSE_LFXO
Generic_nRF51822.menu.lfclk.lfrc=RC Oscillator
Generic_nRF51822.menu.lfclk.lfrc.build.lfclk_flags=-DUSE_LFRC
Generic_nRF51822.menu.lfclk.lfsynt=Synthesized
Generic_nRF51822.menu.lfclk.lfsynt.build.lfclk_flags=-DUSE_LFSYNT
```

Upon flashing I get this error:

```
C:\Users\Marc\AppData\Local\Arduino15\packages\sandeepmistry\tools\openocd\0.10.0-dev.nrf5/bin/openocd.exe -d2 -f interface/{upload.protocol}.cfg -c {upload.setup_command} -f target/nrf51.cfg -c program {{C:\Users\Marc\AppData\Local\Temp\arduino_build_715734/iBeacon.ino.hex}} verify reset; shutdown; 
Open On-Chip Debugger 0.10.0-dev-00254-g696fc0a (2016-04-10-10:13)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
embedded:startup.tcl:60: Error: Can't find interface/upload.protocol.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 60
Der ausgewählte serielle Port at file "embedded:startup.tcl", line 60
 ist nicht vorhanden oder das Board ist nicht angeschlossen
```
unless I comment out this two lines:

```
#Generic_nRF51822.upload.protocol=stlink-v2
#Generic_nRF51822.upload.setup_command=transport select hla_swd;
```

In my boards.txt

I tried this too:

```
Generic_nRF51822.upload.protocol=
Generic_nRF51822.upload.setup_command=
```

but got the following result:

```
C:\Users\Marc\AppData\Local\Arduino15\packages\sandeepmistry\tools\openocd\0.10.0-dev.nrf5/bin/openocd.exe -d2 -f interface/.cfg -c -f target/nrf51.cfg -c program {{C:\Users\Marc\AppData\Local\Temp\arduino_build_991973/iBeacon.ino.hex}} verify reset; shutdown; 
Open On-Chip Debugger 0.10.0-dev-00254-g696fc0a (2016-04-10-10:13)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
embedded:startup.tcl:60: Error: Can't find interface/.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 60
Der ausgewählte serielle Port at file "embedded:startup.tcl", line 60
 ist nicht vorhanden oder das Board ist nicht angeschlossen
```

It seems that the parameters are not correctly replaced in the calling of openocd.exe

sandeepmistry, 2018-04-28:
@cimba007 in https://github.com/sandeepmistry/arduino-nRF5/issues/264#issuecomment-380252070

It seems like things are working fine, I'm not sure why you decided to comment out the lines.

closed 2018-04-28 by sandeepmistry

cimba007, 2018-04-30:
I get the error from the start. I manually added the two lined and commented them out to show the error. If I comment them in I get no error but I had to manually add the lines (they are not there from the start).