sandeepmistry/arduino-nRF5 #267: Question BLE400 + NRF51822 + BMP280 sensor + temp_sensor.ino

labajo, 2018-04-14:
I adapted temp_sensor.ino in order to change the sensor and use a I2C sensor. 

https://gist.github.com/labajo/f3a7d279ade485b4046c9de5c7c820f8

I use the BLE400 variants 

```
#define PIN_WIRE_SDA         (0u)
#define PIN_WIRE_SCL         (1u)
``` 
But I have problems with the I2C communication.  I don't know if those pins values are taken into account when the Adafruit library calls Wire.begin(). The other problem is I cannot use Wire.begin(SDA_PIN, SCL_PIN) to call this method.
Does anyone have experience about this?

Thank you very much.

dlabun, 2018-04-14:
The Adafruit library should communicate with the chip using the proper pins. Have you tried a basic sketch that reads a few registers just to see if it's the library or your wiring?

labajo, 2018-04-15:
Thank you.
The problem was the address of the device. The device is a GY BMP280 instead an Adafruit BMP. It has the 0x76 address instead 0x77. In order to fix it, you need to modify Adafruit_BMP280.h.

Now it works 😃 

Thank you very much.

closed 2018-04-15 by labajo