sandeepmistry/arduino-nRF5 #268: nRF51 Generic Code Upload/Build Problem

eren23, 2018-04-17:
Hi,
I've tried to upload code to my generic nRF51 board, but after hex upload finishes i can't see any softdevice or application file on nRFgo Studio. I'm choosing the board, softdevice, and taking output as hex file after that i'm uploading code via nRFGo Stuido but as i mentioned above i can't see any files up here.

eren23, 2018-04-18:
Thank you for your reply, so can i upload my code with JLink just on Arduino IDE like in nRFgo Studio?

eren23, 2018-04-18:
After i upload my code with the instructions 
[image]

dlabun, 2018-04-18:
Did you follow the directions in the readme file to install the JLink driver?

eren23, 2018-04-18:
Yes i've followed it, i can start it over if you don't have any other suggestions for now.

dlabun, 2018-04-18:
From the screen shot it's indicating that the LIBUSB driver is not installed. The only recommendation I have left is to remove anything JLink related and follow the install instructions on this repository... Don't install the Segger drivers or nRF Studio.

eren23, 2018-04-18:
Okay ty i will try your suggestions, i will write back here as soon as i try them.

eren23, 2018-04-19:
[image]
After i upload my code this is the output now, is this seem fine?

eren23, 2018-04-19:
Now i'm trying a basic code to my nRF51 DK to check my PCB is ok but i couldn't get any solutions either. Here is the picture:
[image]
I think i'm failing about the very basics but i can't get it what are those fails, do you have any other ideas about that matter?

dlabun, 2018-04-19:
Did you have the Arduino IDE open while uninstalling / install the LIBUSB driver?

eren23, 2018-04-20:
I can't say anything for sure but i'll make sure it closed and try it again.

eren23, 2018-04-20:
I've tried it all but the result is still the same, if you don't mind can you explain what should we do exatcly with zadig?

dlabun, 2018-04-20:
I'm not sure what you are asking. I think at this point you simply have a bunch of programs and drivers conflicting with each other... How you fix this problem I do not know.

sandeepmistry, 2018-05-05:
Closing for now due to lack of activity.

closed 2018-05-05 by sandeepmistry