sandeepmistry/arduino-nRF5 #270: Use This repo to create a BLE or to access the RF transreceiver without the BLE stack.

suveergarg, 2018-04-24:
Hi,
Is it possible to use this repo and with some additional hacks, build a BLE central device? Or conversely..Can we access the NRF transreceiver without using the BLE stack? The end goal is to create a sensor network where nodes will communicate with each other, or the hub.

dlabun, 2018-04-26:
No “hacks” are required to this project, all you would need to do is write a BLE Central library in a similar fashion to Sandeep’s BLE Perpherials library.

I do not believe it’s possible to access the radio without the Nordic soft device.

suveergarg, 2018-04-26:
Thank you ! Are you aware of any libraries that provide this functionality for nrf52832 SoC ? I am new to development on ARM SoC and only by using libraries such as these, I am able to build basic apps. Apart from using the Nordic nrf51 SDK and setting up the ARM development environment, is there another quick way to test the full power of the BLE device through basic apps. 
I considered using MyNewt [https://www.sparkfun.com/products/13990](url) but that is not as user friendly to beginners.[(https://davidgs.com/2018/heres-why-im-giving-up-on-mynewt-for-now/](url)

sandeepmistry, 2018-04-28:
@suveergarg there is nothing I'm aware of now, I suggest you follow https://github.com/sandeepmistry/arduino-nRF5/issues/166 for updates or make your own :)

closed 2018-04-28 by sandeepmistry

aditihilbert, 2018-05-01:
@suveergarg, do try Mynewt. It is user friendly to beginners and there is a whole community on dev@ mailing list and Slack who will reiterate that and provide quick help. The BLE controller performs very well and has lots of config options.