sandeepmistry/arduino-nRF5 #271: Electronut Labs hackaBLE + Bumpy unable to flash in Arduino 1.8.5 (Windows 7 / 64)

w8abz, 2018-04-26:
I did some searching here with no luck finding an answer.  I am using Arduino 1.8.5 on Win7/64, using "Nordic Semiconductor nRF5 Boards by Sandeep Mistry version 0.5.1", with the board target set to "Electronut Labs hackaBLE".  I'm using Electronut labs's Bumpy black magic probe as the programmer.  Softdevice S132 is selected, Low Frequency Clock is set to "Crystal Oscillator".  Programmer is set to "Black Magic Probe (GDB)".  Using a simple Blink sketch to blink PIN_LEDB (which is defined in variant.h).  The code compiles fine, but I am unable to flash the device.  The output in the Arduino IDE bottom frame is as follow:

`Sketch uses 2704 bytes (0%) of program storage space. Maximum is 409600 bytes.
Remote debugging using \\.\COM20
Target voltage: unknown
Available Targets:
No. Att Driver
 1      ARM Cortex-M
Attaching to Remote target
warning: while parsing target memory map (at line 1): Required element <memory> is missing
0x00013d6e in ?? ()
Reading symbols from RGB_Blink.ino.elf...done.
Loading section .text, size 0xa90 lma 0x1c000
Loading section .ARM.exidx, size 0x8 lma 0x1ca90
Loading section .data, size 0x70 lma 0x1ca98
Start address 0x1c590, load size 2824
Transfer rate: 45 KB/sec, 564 bytes/write.
Temporary breakpoint 1 at 0x1c894: file C:\Users\Latitude1\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.5.1\cores\nRF5\main.cpp, line 28.
Starting program: C:\Users\xxxxxxxx\AppData\Local\Temp\arduino_build_700356\RGB_Blink.ino.elf 
warning: while parsing target memory map (at line 1): Required element <memory> is missing`

So it seems I'm missing a memory map somewhere. ???
Or is it possible I need to update the firmware on the Bumpy?  (It worked fine for updating the bootloader on a Bluey...)

When the output gets to the above, the Bumpy programmer has the red and yellow LEDs blinking alternately, about 150 ms on-time each.  I've checked that I'm choosing the BMP port and not the UART port on the Bumpy, and the wiring from the Bumpy to the hackaBLE is correct.

Any help appreciated!
Dale

dlabun, 2018-04-26:
I recommend contacting Electronut for help as their programmer may not be perfectly identical to the BMP support we’ve included. I did noticed in the error message that it says the target voltage is unknown, which may be a sign of communication issues.

sandeepmistry, 2018-04-28:
cc/ @ntavish

ntavish, 2018-04-30:
Hi @w8abz, I tested hackable with bumpy just now, seems to work for me, can you verify that you are powering it correctly(Vdd pin on SWD connector takes 3.3v max) and the SWD connections?

Bumpy doesn't measure target voltage, so 'target voltage unknown' is okay.

w8abz, 2018-05-01:
@ntavish, thank you.  I saw in the Bumpy instructions that it does not return voltage.  I'm fairly certain I've damaged my Bumpy -- it no longer finds Bluey when trying a scan from arm-none-...   So I've ordered a replacement, plus another hackaBLE + Bumpy + pogopins + etc. kit.  I'll be more careful with connecting -- probably solder some pins on the hacakBLE instead of just holding on the pogo pins.  Thanks all for the help!   Dale

closed 2018-05-01 by sandeepmistry