sandeepmistry/arduino-nRF5 #272: Set wrong low freqeuncy clock, bricked?

junglejim8, 2018-05-07:
I followed the procedure for a generic nrf52. But I didnt notice the setting for Low frequency clock. It was defaulted to crystal oscillator. My custom board does not have a crystal. 

The original flash of softdevice went fine, the StlinkV2 blinked a lot and the arduino IDE showed it was a success.

But now I can't do anything. Even reflash the bootloader. It keeps saying
"Error: init mode failed (unable to connect to the target)"

Is this possibly because the chip wont work now that it has no clock?

Thanks

dlabun, 2018-05-07:
You should still be able to communicate with it via the programmer... Have you tried completely disconnecting the programmer and then trying to flash again?

dlabun, 2018-05-20:
No response from Op, Closing

closed 2018-05-20 by dlabun