sandeepmistry/arduino-nRF5 #274: Support for RuuviTag board

giowild, 2018-05-17:
Hi!
Do you plan to add RuuviTag support? It's a Nordic 52832 chip!

It would be great to be able to correctly compile the .hex file with Arduino IDE, in order to upload it over OTA

These are references:
website: https://ruuvi.com/
pin configuration: https://lab.ruuvi.com/pinout/
Firmware development with examples: https://github.com/ruuvi/ruuvitag_fw
linker script example: https://github.com/ruuvi/ruuvitag_fw/tree/master/ruuvi_examples/ruuvi_firmware/ruuvitag_b/s132/armgcc

dlabun, 2018-05-17:
As far as I know there's nobody working on it, but you can certainly submit a PR to add it.

giowild, 2018-05-18:
Hi @dlabun, I'll give try to implement it, but I can't understand the exact content of variant.h and variant.cpp inside each variant folder...

What would be the exact way of mapping them with https://lab.ruuvi.com/pinout/ ?

Do you have there a board integration guide?

dlabun, 2018-05-20:
You can find more information on how the Arduino platform works here:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification

After reading the specs it should make sense on how all of the files work.