sandeepmistry/arduino-nRF5 #279: UBLOX B112

tcpipchip, 2018-05-24:
I have succesfully execute some BLE examples on UBLOX B112 (NRF52822) on Arduino.

But...when use the Serial.print.... i can´t see data on TXD of NRF52822....

Is that a bug ? And where is the doc of PINOUT ARDUINO to acess the NRF52822 gpios ?

Thank you!!!!

dlabun, 2018-05-24:
It's unlikely that it's a bug. Since none of the UBlox modules are currently supported by this project you would need to post all of your variant information so we can see what may be wrong.

tcpipchip, 2018-05-24:
Hi, do you mean Hardware Pinout ? Schematics ?

dlabun, 2018-05-24:
No, your modified variant.h, and variant.cpp files... The programmer information would also be helpful.

tcpipchip, 2018-05-25:
I finished 98% of conversion to work to NINA B112, i use brute force to find out pinout config.

What i can believe is, why i found the following pinout to NINA

#define PIN_A0               (8)  //OK
#define PIN_A1               (7)  //OK
#define PIN_A2               (9)  //OK
#define PIN_A3               (5)  //OK
#define PIN_A4               (4)  //OK
#define PIN_A5               (3)  //OK

and the TAIDA CENTURY uses

#define PIN_A0               (25)
#define PIN_A1               (26)
#define PIN_A2               (27)
#define PIN_A3               (28)
#define PIN_A4               (29)
#define PIN_A5               (30)

Both uses the same processor!

Why are they different ?

Miguel

dlabun, 2018-05-25:
It all depends on what the module manufacturer does for a PCB layout. There's no need to use brute force for anything, the UBlox datasheet lists the mapping of all the pin functions.

tcpipchip, 2018-05-25:
But why only my config works ? Pinout setup!

Enviada do meu iPhone

dlabun, 2018-05-25:
I'm sorry, but I honestly do not understand what you are trying to say. 

You cannot compare a Ublox module and a Taida Century module because they are completely different.

tcpipchip, 2018-05-25:
But they are the  same processor!

Maybe there is some that i am not understanding! How you extracted from  taida ptocessor datasheet the correspondent file config to Arduino ?

Miguel

Enviada do meu iPhone

dlabun, 2018-05-26:
While they have the same processor, these are complete modules. That means it is up to the module manufacturer on how they breakout the nRF pins to the module pins. There is also the fact that since these modules are not attached to a board you can assign whatever Arduino pin numbers you want to the Ublox pin numbers.

closed 2018-05-27 by dlabun