sandeepmistry/arduino-nRF5 #282: Use Maple Mini to program nrf51822

francesco98, 2018-06-04:
Hi,

Is it possibile to program nrf51822 using Maple Mini + USB to TTL converter?

Thanks,
Regards.

dlabun, 2018-06-04:
No, you need a programmer capable of ARM SWD.

francesco98, 2018-06-04:
Thanks for your support, but in this article, that point to this repository, it seems possibile.
[http://www.rogerclark.net/arduino-on-the-nrf51822-bluetooth-low-energy-microcontroller/](http://www.rogerclark.net/arduino-on-the-nrf51822-bluetooth-low-energy-microcontroller/)

dlabun, 2018-06-04:
Your original question was about a USB to TTL converter which are not capable of programming an nRF chip.

If you want to program the Black Magic Firmware onto a board and use it as a SWD programmer, then yes in theory it should work. Others have done this but be aware there have been issues related to the BMP firmware itself.

francesco98, 2018-06-04:
Ok, I have asked if it is possibile because I don't want to buy the nRF51822 Development Board. Can you suggest me a programmer cheaper and easier to use? On Internet I just found ST-Link v2, what about this?

Thanks.

dlabun, 2018-06-04:
Formally we support the Seger JLink, ST-Link v2 and the Black Magic Probe. I'm sure others may work with some tweaking, but it'd be difficult for us to provide any sort of tech support for it.

francesco98, 2018-06-04:
Ok, thank you very much for your support.

closed 2018-06-04 by dlabun