sandeepmistry/arduino-nRF5 #283: Need to change pin for Redbearlab nrf51822 board for spi communication

Devanshi15, 2018-06-05:
I am using Redbearlab nrf51822 board in arduino IDE , default pin configuration compatible for arduino is as follows:

 SS- 10  // P0.14 for nRF51422 board PCA10028
SCK- 16  //P0.25
MISO- 17 ///P0.22
MOSI - 18//P0.30

I get the data by this pin configuration but I want to change this pin configuration to :

SS- 10 //P0.14 
SCK-  //P0.20     
MISO-17//P0.22
MOSI-4//P0.21

How do I change my default pin configuration to the one I need ??

dlabun, 2018-06-05:
You need to change the pin numbers in variant.cpp and variant.h.

Devanshi15, 2018-06-05:
I am confused regarding variant.h and variant.cpp files from which folder do i need to change. 
Also , when i clone this into hardware folder and change Rebearlab nrf51822- variant.h and variant.cpp files there is no change in pins in hardware.

I have doubt that does it take default library as it is from Arduino IDE- File-Preference - Additional Board Manager URL - https://sandeepmistry.github.io/arduino-nRF5/package_nRF5_boards_index.json

dlabun, 2018-06-05:
You can do it two ways... The simplest way is to modify the files in the folder for the board you're using (in this case the Redbear). For the more advanced user, you can create a new variant with a different name and pin out as you desire.

Devanshi15, 2018-06-06:
ok so I have to clone this to hardware folder of documents/arduino or programfilesx86/arduino ? and there I need to change the pins in variant.h of RedbearLab_nrf51822 folder-- variant.h and variant.cpp 
under const uint32_t g_ADigitalPinMap[] = {

----
//MISO, SCK, MOSI

22
20
21 

right? and comment wherever this pin is used additionally for different purpose?

In variant.cpp

#define PIN_SPI_MISO    (22)
#define PIN_SPI_MOSI    (21)
#define PIN_SPI_SCK       (20)

Am i correct ? Are there any additional steps i need to do?

Devanshi15, 2018-06-06:
@dlabun when i clone this into hardware and change the variants I dont see any change in pin config when I connect the device and test it ?

mindforger, 2018-06-06:
@Devanshi15 what is the reason to change the SCK Pin? 

1. Just a case of special wiring or 
2. is the Pinning in the Variants Folder for your board just simply not working?

https://github.com/sandeepmistry/arduino-nRF5/tree/master/variants/RedBearLab_nRF51822
>// MISO, SCK, MOSI
  22,
  25,
30,

i don't have that board and no schematics, so i don't understand why you want to change them

if it were 1. you should only clone/download this repo, change the files and compile it for yourself 
if it were 2. you could either clone the repo, change the file and send a pullrequest or maybe a dev can fix that peorbably wrong file for you :)

Devanshi15, 2018-06-06:
I changed the files after clone, and I want to change the pins because my hardware PCB demands that. And the variant is working fine when I configure it with default pins mentioned. 

But after changing variants I don't find any change .. it works with default pins only. It might be that I am doing some step wrong .... or missing it. 
Do I need to make any change in my arduino code to support the change in variants??

dlabun, 2018-06-06:
Since this not a problem related to this repository but rather an attempt to customize a board I am going to close this issue. I would recommend reading the Arduino variant file specs or if you need more in depth troubleshooting should ask on the Arudino forums.

closed 2018-06-06 by dlabun