sandeepmistry/arduino-nRF5 #284: Use P0.00 as ADC Pin

francesco98, 2018-06-06:
Hi,

I'm reading http://infocenter.nordicsemi.com/pdf/nRF51822_PS_v3.3.pdf and in section 2.2 it seems possibile to use P0.00 as Analog Input. Is it possibile? Which is its name in code?

Thanks

francesco98, 2018-06-07:
Thanks for your answer.
I'm using CJMCU-8223 that is a board with nRF51822 and LIS3DH (accelerometer). Unfortunally, on this board the only ADC pin is P0.00 on which I want connect the Pulse Sensor ([https://pulsesensor.com/](https://pulsesensor.com/)). Can you help me to use this pin?

Thank you

closed 2018-07-03 by dlabun

yogeshwaran774, 2019-02-18:
Hi, i am using cj-mcu 8223 and i have the nrfsdk but i don't have datasheet of cjmcu. i googled it but still can't find any resources. i uploaded some example codes by compiling in keil u5 but no results after uploading. so can anyone provide me a link of cjmcu data sheet and sample codes.

sinaban, 2021-01-13:
this is a terrible board. there is no data and even no reset pin, reset pin for nrf52811 is p0.21, unfortunately, we have not pinout for it in PCB,