sandeepmistry/arduino-nRF5 #287: Arduino IDE cannot upload the program

scientistnobee, 2018-06-24:
Hi everyone, I am working on ID107HR device which has a nRF51. I was able to upload the code, If I chose softdevice as "none" in the "tools" "softdevice" options. However, But if I chose "s130", Arduino IDE is struck while uploading the code to the nRF device. 

The Arduino debug message is as following:

`Build options changed, rebuilding all
Sketch uses 9096 bytes (6%) of program storage space. Maximum is 151552 bytes.
Remote debugging using \\.\COM15
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      Nordic nRF51
Attaching to Remote target
0x0000102a in ?? ()
Reading symbols from Lux_sensor.ino.elf...done.
Loading section .text, size 0x2388 lma 0x1b000
Loading section .ARM.exidx, size 0x8 lma 0x1d388
Loading section .data, size 0x80 lma 0x1d390
Start address 0x1b7a0, load size 9232
Transfer rate: 15 KB/sec, 769 bytes/write.
Temporary breakpoint 1 at 0x1be6a: file C:\Users\nobee\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.5.1\cores\nRF5\main.cpp, line 29.
Starting program: C:\Users\nobee\AppData\Local\Temp\arduino_build_972439\Lux_sensor.ino.elf 
Note: automatically using hardware breakpoints for read-only addresses.`

So far  I was using "none" as softdevice option without any problem. But when I would like to work with BLEPeripheral, if I include the line #include <BLEPeripheral.h> in the sketch, I get the following error. C:\Users\nobee\Documents\Arduino\libraries\BLEPeripheral\src/nRF51822.h:11:25: fatal error: ble_gatts.h: No such file or directory. If I chose softdevice as "s130" in the "tools" "softdevice" options, the error goes away. So this is forcing me, not to use "none" as a softdevice option.

Does anyone encountered the above problem? Please let me know, how to troubleshoot this problem.

dlabun, 2018-06-24:
Did you flash the soft device prior to uploading your code?

scientistnobee, 2018-06-24:
Yes, I flashed soft device already and my code to control the screen, accelerometer is working without a problem when I chose "none" as the option. But when I want to use BLEPeripheral, the code is not compiling with "none" as softdevice option. On the otherhand, if I choose softdevice, code is not getting uploaded.

scientistnobee, 2018-06-24:
I am using black magic probe as a programmer to upload the programs. And my computer is running Windows 8.1, if these details are of any use.

dlabun, 2018-06-24:
The BLEPeripheral repository will not build without the soft device selected.

There have been similar reported issues with the Black Magic Probe. I do not own one of those programmers so I can’t help troubleshoot it but I believe most of the people reporting issues had to update their firmware.

scientistnobee, 2018-06-24:
Thanks for the help. I will try to update my firmware and let you know. I will be away from my hardware for a week. Thanks again.

closed 2018-07-03 by dlabun

scientistnobee, 2018-07-09:
Dear @dlabun I updated my firmware today after a lot of help from BMP community. Still the same problem persists. Do you have suggestions?

gsmcmullin, 2018-07-09:
Looking at the output, it appears the Black Magic probe is behaving correctly and target flashing succeeds.  The could be confirmed by sending GDB the `compare-sections` command after `load`.

The target is running but not hitting the expected breakpoint.  I suspect there is a problem with the generated elf file, but I don't have hardware try reproduce the problem

dlabun, 2018-07-09:
Please do me a favor and reflash the SD132 and post that debug message, followed by flashing your code and post that debug message as well.

scientistnobee, 2018-07-09:
Dear @dlabun I always connect the watch to the USB port through its charging cable. So its powered from laptop. I first uploaded the softdevice using instructions given on this repository. Then I went to add Mark's board to get the id107 as one of the variant. https://github.com/micooke/arduino-nRF5-smartwatches
Now I can't see flash s130 option anymore on Arduino IDE. Tomorrow, I will start from scratch and see if I can solve this. Thanks for your patience.